flipperdevices/flipperzero-firmware #1481: Can't compile using protobuf==3.20.1

ClusterM, 2022-07-28:
### Describe the bug.

There is protobuf 3.20.1 in `scripts/requirements.txt` but firmware can't be compiled using this version.

### Reproduction

* Checkout https://github.com/flipperdevices/flipperzero-firmware.git
* Run `pip3 install -r scripts/requirements.txt`
* Run ./fbt

### Logs

```Text
Traceback (most recent call last):
  File "/home/cluster/flipperzero-firmware/lib/nanopb/generator/nanopb_generator.py", line 51, in <module>
    from .proto import nanopb_pb2
ImportError: attempted relative import with no known parent package

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/cluster/flipperzero-firmware/lib/nanopb/generator/nanopb_generator.py", line 72, in <module>
    import proto.nanopb_pb2 as nanopb_pb2
  File "/home/cluster/flipperzero-firmware/lib/nanopb/generator/proto/nanopb_pb2.py", line 10, in <module>
    from google.protobuf import symbol_database as _symbol_database
  File "/home/cluster/.local/lib/python3.9/site-packages/google/protobuf/symbol_database.py", line 189, in <module>
    _DEFAULT = SymbolDatabase(pool=descriptor_pool.Default())
AttributeError: module 'google.protobuf.descriptor_pool' has no attribute 'Default'
scons: *** [build/f7-firmware-DC/assets/compiled/application.pb.c] Error 1
********** ERRORS **********
Failed building build/f7-firmware-DC/assets/compiled/application.pb.c: Error 1
```

### Anything else?

Installation of latest protobuf solves this problem:
```
$ pip3 install --upgrade protobuf
Collecting protobuf
  Using cached protobuf-4.21.4-cp37-abi3-manylinux2014_x86_64.whl (408 kB)
Installing collected packages: protobuf
  Attempting uninstall: protobuf
    Found existing installation: protobuf 3.20.1
    Uninstalling protobuf-3.20.1:
      Successfully uninstalled protobuf-3.20.1
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
grpcio-tools 1.47.0 requires protobuf<4.0dev,>=3.12.0, but you have protobuf 4.21.4 which is incompatible.
```
There is error in the last line but at least firmware can be compiled now.

hedger, 2022-07-28:
Which distro are you running fbt on?
Also fbt and Python scripts it calls should use Python interpreter distributed with fbt's toolchain. Do you have it in `toolchain` foler?

ClusterM, 2022-07-28:
> Which distro are you running fbt on?

Linux ovh.clusterrr.com 5.10.0-16-cloud-amd64 #1 SMP Debian 5.10.127-1 (2022-06-30) x86_64 GNU/Linux

> Also fbt and Python scripts it calls should use Python interpreter distributed with fbt's toolchain. Do you have it in `toolchain` foler?

Yes, I have it but I used installed Python 3.9.2.

improving-rigmarole, 2022-07-28:
hi,
I just followed [this readme](https://github.com/flipperdevices/flipperzero-firmware/blob/dev/documentation/fbt.md),
ran `fbt.cmd`, and the exact same error occurred. I'm using Windows10.
From the readme, it seems to indicate it's needed to have python, GNU arm toolchain, and OpenOCD installed and in path, but it seems that fbt itself is supposed to install everything on its own (or at least use its own python) ?

the first time I tried was with release branch,
and I just retried with dev, and this time it installed its toolchain.

skotopes, 2022-07-28:
@drunkbatya can you take a look?

drunkbatya, 2022-07-29:
I’ve reproduced this bug and it’s really strange because fbt’s Python has the very same versions of python libs. Gotta need some time for debugging, I’ll catch you up on it. Have you tried compiling it with ./fbt’s Python?

ClusterM, 2022-07-29:
> Have you tried compiling it with ./fbt’s python?

No, I have not.

skotopes, 2022-09-06:
Please check latest release and reopen issue if problem persist

closed 2022-09-06 by skotopes