flipperdevices/flipperzero-firmware #1482: Regression of tag ID (NFC-A / Mifare Ultra) in 0.63.2-rc

UberGuidoZ, 2022-07-28:
### Describe the enhancement you're suggesting.

In a previous firmware (0.62.1) I was able to scan an NFC-A/Mifare Ultra card and have it instantly display data

[image]

Since upgrading to 0.63.2-rc, it now seems to hang at the following (5 min and counting)

[image]

After some [discussion in Discord](https://discord.com/channels/@me/994304126974435338/1002266271301255231), it was determined this is likely not "expected" behavior and regression in features. Since this particular NFC is passworded/encrypted, it is never able to fully read it, which is what it appears to be doing (and never finishing.) Because of this, not even the ID of the tag is known.

A suggestion was to add a "fixed retry count" added so it could gracefully fail and still ID the tag.

LOG from 0.62.1 (start NFC ->Read, takes 1-2 seconds.)
```
>: log
Press CTRL+C to stop...
3335679 [I][LoaderSrv]: Starting: NFC
3335687 [I][AnimationManager]: Unload animation 'L1_Waves_128x50'
3337919 [D][DolphinState]: icounter 6, butthurt 0
3338120 [D][DolphinState]: icounter 9, butthurt 0
```

LOG from 0.63.2-rc (start NFC ->Read, takes 1-2 seconds for repeating log, left running 2 mins.)
```
>: log
Press CTRL+C to stop...
140220 [I][LoaderSrv]: Starting: NFC
140228 [I][AnimationManager]: Unload animation 'L1_Recording_128x51'
143757 [D][DolphinState]: icounter 215, butthurt 3
143948 [I][NfcWorker]: Mifare Ultralight / NTAG detected
144120 [D][MfUltralight]: Reading version
144125 [D][MfUltralight]: Failed reading version
144176 [D][MfUltralight]: Reading pages 41 - 44
144181 [D][MfUltralight]: Failed to read pages 41 - 44
144232 [D][MfUltralight]: Reading pages 0 - 3
144238 [D][MfUltralight]: Reading pages 4 - 7
144244 [D][MfUltralight]: Failed to read pages 4 - 7
144512 [I][NfcWorker]: Mifare Ultralight / NTAG detected
144682 [D][MfUltralight]: Reading version
144687 [D][MfUltralight]: Failed reading version
144746 [D][MfUltralight]: Reading pages 41 - 44
144754 [D][MfUltralight]: Failed to read pages 41 - 44
144805 [D][MfUltralight]: Reading pages 0 - 3
144812 [D][MfUltralight]: Reading pages 4 - 7
144818 [D][MfUltralight]: Failed to read pages 4 - 7
145086 [I][NfcWorker]: Mifare Ultralight / NTAG detected
```

### Anything else?

Since it appears to be failing instantly (1-2 seconds), but simply staying in a loop, the fixed retry count should allow it to gracefully exit the said loop and at least identify the card (with previously included data for info) as it appears to do so, just never gets a chance to show it. Could likely include a brief error message (passworded/encrypted/etc.).

UberGuidoZ, 2022-07-29:
Similar issues with other encrypted/passworded cards.

[image]

phraktle, 2022-08-07:
I can confirm this bug, it was able to read the NFC with the previous release firmware.

closed 2022-09-06 by skotopes