flipperdevices/flipperzero-firmware #1494: Alow universal IR for AC and Audio

jaroslavmraz, 2022-07-29:
### Description of the feature you're suggesting.

Hi,

Would it be possible to enable the universal controller for Air Conditioning and Audio devices with brutforce. I have already prepared an ir file for Air Conditioners based on the tv.ir pattern from the assets folder. And I'm happy to share it.

skotopes, 2022-09-06:
@gsurkov is currently working on it. @jaroslavmraz your file will be quite useful.

jaroslavmraz, 2022-09-07:
Happ for it :)

Last and more uptudate version made by comunity are here. Comnuity random adding new irs find it here @skotopes 

https://github.com/UberGuidoZ/Flipper/tree/main/Infrared

UberGuidoZ, 2022-09-08:
Also note: There have been custom-developed remote GUI too that supports the needs (like Mode) for AC. Others for Speakers, Fan, etc.

gsurkov, 2022-10-14:
A/C universal remote is available now, and there is a [manual](https://github.com/flipperdevices/flipperzero-firmware/blob/dev/documentation/UniversalRemotes.md#air-conditioners) for adding new signals.
As for the Audio, I think the library linked above would suffice to start, will add it soon.

UberGuidoZ, 2022-10-17:
Sounds great. We are trying to keep these files up to date, pulling from the IRDB. 2-3 people are adding while another few are verifying functionality and duplicate/corruption removal. There are linked scripts from @darmiel used for this.

gsurkov, 2022-11-08:
The Audio universal remote is done as well. However, the aforementioned database could not be used as it lacked the buttons we wanted to include. 
The [manual](https://github.com/flipperdevices/flipperzero-firmware/blob/dev/documentation/UniversalRemotes.md) was updated with the Audio-specific info, too.

closed 2022-11-08 by gsurkov