flipperdevices/flipperzero-firmware #1495: Python scripts do not find python libraries installed in the new toolchain

Kingwad, 2022-07-29:
### Describe the bug.

The new toolchain folder includes several python libraries, for example pyserial, but the python scripts, for example storage.py, still require libraries to be installed in the system.

### Reproduction

remove pyserial library from the os installed Python libraries
Make sure the toolchain folder is up to date via fbt
Confirm that the toolchain contains the serial package (toolchain/x86_64-linux/python/lib/python3.9/site-packages/serial)
Run storage.py
See that it cannot find the serial package

### Logs

```Text
>: python scripts/storage.py 
Traceback (most recent call last):
  File "/home/user/Development/flipperzero-firmware/scripts/storage.py", line 4, in <module>
    from flipper.storage import FlipperStorage
  File "/home/user/Development/flipperzero-firmware/scripts/flipper/storage.py", line 3, in <module>
    import serial
ModuleNotFoundError: No module named 'serial'
```

hedger, 2022-07-29:
Toolchain and supplemental environment are set up by additional scripts. Soon there will be a way to spawn a shell with all toolchain's tools in it, so what you described will no longer be an issue.

skotopes, 2022-09-06:
Check `./scripts/toolchain/fbtenv.sh` and reopen issue if problem persist.

closed 2022-09-06 by skotopes