flipperdevices/flipperzero-firmware #1498: Add NFC Toys Keys Algorythm for NFC-A Skylanders

V0lk3n, 2022-07-29:
### Describe the enhancement you're suggesting.

Add Keys Algorythm for NFC-A Skylanders.

Skylanders Toys is Mifare Classic Standard using SKA01.

Actually reading the toys with the flipper only read the UID part.

 @[image]

Using this UID, i was able to retrieve the 16 keys using ```tnp3xxx.py``` script from NFC Toys that you can find <a href="https://nfc.toys/interop-sky.html">here</a>

Once I got the 16 keys from the script, I add them to "mf_classic_dict_user.nfc" i'm not able to retrieve the data of Skylanders using the flipper. Due to the keys algorythm not present for SKA01 at what i've understand on a discord discussion.

But to confirm that I'm on the right track, I've downloaded on my Phone the tool "Mifare Classic Tool" that you can find <a href="https://github.com/ikarus23/MifareClassicTool">here</a>, I've scanned my Skylanders and it returned "unrecognized tag". So i've addedmy 16 keys to the "std.keys" file, which is the same as "mf_classic_dict_user.nfc". Re-reading my skylanders recognized it this time and found the Sector data.

[image]
 

### Anything else?

I link it my previous issue, as it was the beginning of my story with those skylanders. May can help : https://github.com/flipperdevices/flipperzero-firmware/issues/1469

Additionnaly https://nfc.toys/docs.html should be a good ressource.

V0lk3n, 2022-07-29:
Well sorry for the LAAAARGE picture, wasnt intended...

V0lk3n, 2022-07-29:
@gornekich

V0lk3n, 2022-07-29:
Seem that this communauty is soooo FAST and efficiant :) @equipter just made a pull request for it https://github.com/flipperdevices/flipperzero-firmware/pull/1497

gornekich, 2022-08-01:
Hi @V0lk3n , thanks for information! Also thanks @equipter for PR!
We will merge this PR soon, when Skylander figures comes to office to verify.
I never heard about this nfc tags. Is Skylander popular? As I understand you have to by specific reader to add your Skylander figures to the game, is that right?

V0lk3n, 2022-08-01:
> Hi @V0lk3n , thanks for information! Also thanks @equipter for PR!
> We will merge this PR soon, when Skylander figures comes to office to verify.
> I never heard about this nfc tags. Is Skylander [...]

Hi! Your welcome :).

Skylanders is a game, with a plateform that once you put your figure on it, they are invoked in game ^^

I was playing on ps3 but i know there is for xbox and more

I didnt really play anymore, it was like in 2009 something like this. But once i received my flipper i was trigger by that ^^

So yes there is a plateform, will take a picture once i reech home, also if you need i can try on my own skylanders :)

Also i wonder if we can trick something to made any reader make the work but i dont know, i never touched this type of stuff, started since i got my flipper to learn ^^ but if i fond something more i add it here :)

V0lk3n, 2022-08-01:
https://skylanders.fandom.com/en/wiki/Portal_of_Power

I think this is better than only a picture xD

V0lk3n, 2022-08-08:
Tested your PR and it work like a charm :D any idea when you will merge their algo?

V0lk3n, 2022-08-18:
> Tested your PR and it work like a charm :D any idea when you will merge their algo?

@gornekich

gornekich, 2022-08-18:
Hi @V0lk3n . We decided that we don't want to support these cards in official firmware. Now you can add extracted keys to user dictionary. Algorithm support sounds great for external plugin for Flipper, so we will wait until plugins support is done.

V0lk3n, 2022-08-18:
Ah.

closed 2022-08-18 by V0lk3n