flipperdevices/flipperzero-firmware #1499: settings_user

jeremyyablan, 2022-07-29:
### Description of the feature you're suggesting.

Please make the 
subghz/assets/settings_user 
Into a text file settings_user.txt

I need to modify it from the iPhone when mobile but currently have to wait until I get back to the office. Sometimes you just need a new frequency. 😁👍🏼

### Anything else?

You guys are doing awesome! Thank you.

cixio, 2022-08-12:
I think it would be better to allow a manually entered frequency without using the App.

If you want to edit more details (like hopper frequencies) this sould be adressed to the Flipper iOS App: https://github.com/flipperdevices/Flipper-iOS-App

skotopes, 2022-09-06:
That is something that you need to report in mobile repository.

closed 2022-09-06 by skotopes