flipperdevices/flipperzero-firmware #1500: EM4100 RFID issue: Emulation does not work

swittenberger, 2022-07-30:
### Describe the bug.

Hello, I've had 3 different EM4100 tags that I could copy without any issues, but when emulating the tag, the reader doesn't detect anything. 

I've been told on discord that a reboot might help, but it didn't. What other info can I give? 
[image]

### Reproduction

1. Read EM4100 tag
2. Emulate EM4100 tag
3. Try to open the door with flipper
4. Notice it doesn't work.

bettse, 2022-07-30:
I don't have first hand experience, but I've seen at least 3 other people with the same problem

Astrrra, 2022-07-31:
I haven't been able to replicate this issue.

https://drive.google.com/file/d/1fyJLHXRghv-VP-zwfWvneXZ-uCqj5pWQ/view?usp=drivesdk

As I asked in the reddit thread, can you please attach a video of you trying to emulate the card? This sounds like you may be doing something wrong.

swittenberger, 2022-07-31:
Hi, here is a video of the flipper not being able to emulate the RFID tag. https://drive.google.com/file/d/1clRwR54LT4bUGVUxHWEE8aykKtJO0XuV/view?usp=sharing

I will also check with a 2nd flipper as you did above, but that is a totally different case.

Catman155, 2022-07-31:
I can successfully use the EM4100 emulation on this product: 
https://www.surepetcare.com/en-gb/pet-doors/microchip-cat-flap-connect

Kritkatten, 2022-07-31:
Filmed the process as requested: [link to onedrivre](https://drive.google.com/file/d/1Rsca-HAfT6DrBxRyXwSf_PAS5yOF4d2z/view?usp=drivesdk)

swittenberger, 2022-07-31:
A small update:

I've got my hands on another Flipper and was able to read from the flipper. (One emulates, the other reads) This works. 
I've tested emulation with beta firmware as well (just in case) and I could still reproduce the issue. 

The manufacturer in my case is electra.ro 
https://electra.ro/ro/produse/videointerfoane-si-interfoane/gama-pass-digital/p4s-a91i

WhiteThePanda, 2022-07-31:
Hello, my issue is pretty straightforward : 
Emulation works for the front door of the building, mailbox and other doors it is supposed to open (systematically):
https://user-images.githubusercontent.com/61516945/182038788-6b4f807d-abf3-420f-8743-67f710db17f9.mp4

and doesn't work for the door of my own appartment:
https://user-images.githubusercontent.com/61516945/182038791-9f130302-9a54-4a26-adee-ca7a54a4d012.mp4

The same key is supposed to open both.
I also tried removing the silicon case.

DrZlo13, 2022-08-01:
OK, this may be some kind of emulator protection method, or a hardware problem with concrete reader.
I will return to this topic when I finish rewriting LF-RFID core.

swittenberger, 2022-08-01:
Another update:
I've tested another tag, on another reader of different make and it works without issues. I do think it might be a security thing then on the electra one.

0x77dev, 2022-08-02:
Hey there!

I'm trying to read tag A and write it to tag B
But it only works when I try to write dumped tag A to tag A and tab B to B
and not tag A to B or B to A

Tag A and B are EM4100

[Here is a video](https://www.youtube.com/watch?v=RGgOR_UNZNM)

trupus, 2022-08-31:
Hey guys, I had also troubles emulating the tag on an electra reader.

> The manufacturer in my case is electra.ro https://electra.ro/ro/produse/videointerfoane-si-interfoane/gama-pass-digital/p4s-a91i

Not sure if it's the exact same model as in this link, but it looks very similar to mine.

I also captured the [RAW RFID data](https://drive.google.com/drive/folders/1WnuFCMMm8dlRy8bc0wrMr3jZOFenb48e?usp=sharing). Hope it helps.

skotopes, 2022-09-06:
Please check latest release and reopen issue if problem persist.

swittenberger, 2022-09-06:
@skotopes I just tested on Dev build d1c79a83. I also tested on the stable release 0.66.1. Door still won't open. 
I have reread the rfid just in case for this test, have not used the saved one (though I did try that as well)

https://drive.google.com/file/d/1l_reeBhEbq9uMtiGM9a-h2UJR6l6ayJF/view?usp=sharing

trupus, 2022-09-06:
I had also the same problem with the latest rc today. But with a different reader this time (Flipper identified it as an EM4100 tag as well). Will try tomorrow again with the stable release.

Maybe also worth mentioning, my office has 2 doors, which you can open with the same key. It's a coworking space. The first door is "shared", but the second one can only be opened with my key. I can open the first door (and any "shared" reader in the office) with the Flipper, but the second reader is just refusing to accept the emulated key with the flipper.

DrZlo13, 2022-09-07:
OK, it's time to return to this topic. I will try to reset the protocol encoder if field is gone for ?? (10?) clocks.
Also, it will help if someone points out the problematic reader model.

Kritkatten, 2022-09-07:
Still doesn't work. I saw new data with the last update though. Picture:
[image]

trupus, 2022-09-07:
> Also, it will help if someone points out the problematic reader model.

At this moment I encountered 2 different readers in 2 different countries that share the described problem. I found the manufacturer for one of them: [official page](https://www.primion.de/en/products/access-control-systems/electronic-locks-and-cylinders/pkt-door-handle/), [datasheet](https://www.primion.de/fileadmin/media/Downloads/EN_Data_sheets/pKT_Door_handle_en.pdf)

swittenberger, 2022-09-07:
@DrZlo13 The reader described in this bug is an Electra, as can be found in the link above. What other information would you require?

Montecri, 2022-09-14:
Got the same issue with a different brand of reader. In my case, I even tried proxmark3 cloning to a blank key (besides flipper emulation) and the reader didn't detect. Read somewhere that some readers will try to detect if the key is writable and deny access.

d3xt3r01, 2022-11-27:
I can confirm the electra keyfob issue. Same here using the latest firmware.

LowSkillDeveloper, 2022-12-13:
I have a "mizip" key for a coffee machine. When I try to read it, this is what I get. 

[image]

I don't know if this information is correct or not.  Perhaps the key does not have its own id. And inside the key there is other information. After all, this key has a balance that is displayed when taking coffee.

Ferferite, 2023-01-13:
Any update on the Electra readers? Could try sniffing with a proxmark3 if that provides any useful unfo

Ferferite, 2023-01-14:
After a bit more digging I found some stuff. First, electra.ro is associated with electra-automation.at and they developed their own rfid solution "for maximum security". And according to [this thread](https://forum.flipperzero.one/t/electra-intercom/6368/14), [this ](https://www.nxp.com/docs/en/data-sheet/HTSICH56_48_SDS.pdf)should be the datasheet for the readers. 
All tags made after 2010, they added an extra bit to the standard 125khz cards and most hardware designed to dumbly read a bit stream won't see it right. [Source](https://discord.com/channels/740930220399525928/954422698879098990/999467579737378887)

just compiled some info I could find online, didn't have more time to look into it yet

R1DEN, 2023-06-26:
@DrZlo13 hey there, it seems like a working PoC is already there https://forum.flipperzero.one/t/electra-intercom/6368/65
however it would probably be better to add another protocol to the list.
If you could point to some contribution guide for such issues, I could probably open some kind of a PR, however my C skills are really rusty...

R1DEN, 2023-09-30:
@DrZlo13 any chance we could see this in the firmware any time soon?
Feels like the issue is somewhat researched and the solution is a couple of steps away for experienced contributors. :pray:

skotopes, 2023-10-12:
@R1DEN more like everyone is busy with other tasks. We'll come back eventually to this one.

R1DEN, 2023-10-13:
@skotopes thank you for the reply. Just hoping it will be sooner rather than later as this will really help for Romanian and Moldovan users, we have these intercoms almost everywhere in new apartment blocks.

dontbug, 2023-10-13:
We are waiting, we are really waiting. Thank you in advance

gherman22, 2023-12-21:
Any luck to emulating the tag on an electra reader for anyone?

R1DEN, 2023-12-21:
@gherman22 only with that "hacky" way from one of my above posts. Hope a proper way will be incorporated in the firmware some day...

gherman22, 2023-12-21:
Thank you for your answer, i tried to fallow the steps but i got lost. Mabe we will have a easier way or a video with the steps.

skotopes, 2024-01-03:
@DrZlo13 ping

dontbug, 2024-01-03:
@DrZlo13 ping

closed 2024-05-22 by skotopes

d3xt3r01, 2024-05-22:
Can confirm it works!

itrack, 2024-06-20:
Should this function work?

d3xt3r01, 2024-06-20:
Yes. Works, tested.

itrack, 2024-06-20:
I managed to emulate, it works fine but when I tried to write a blank tag it didn't work:

[image]

I mention that this only happens with Electra tags, with other RFID tags it works ok

itrack, 2024-06-20:
it's a bit strange, if I read a tag and write over the same tag it works, if I want to write it on a different one it doesn't work

bettse, 2024-06-20:
> I tried to write a blank tag

t5577? (https://docs.flipper.net/rfid/write-data)

> write over the same tag it works

Generally not a valid test, even for a read-only tag it would appear to be successful because the tag has the same data.

itrack, 2024-06-20:
t5577 and Electra blank tag, on t5577 I managed to write other tags that are not of the Electra type

skotopes, 2024-06-20:
@itrack do I understand right that writing to 5577 fails too? If yes then have you tried to read this 5577 with proxmark?
Some protocols got additional payload validation and may alter data written to 5577 which may cause write fail. May be that is your case?

As of electra blank keyfobs, do you know what exactly is in the keyfob? Do they also use 5577?

VladFlorinIlie, 2024-06-20:
Why do you even bother to write an Electra tag? They are more expanse and harder to come by. As for T5577, I have tested writing them and they are recognized by the reader. I would say that reading an Electra tag and then writing the data on a T5577 should be your best bet on getting a proper tag.

itrack, 2024-06-20:
Hi @VladFlorinIlie ,

I put here a demo in which I try read electra tag and to write to T5577 and another an electra tag:
https://www.youtube.com/shorts/fI8o7a28TzE

It doesn't work.

After to prove that it is a programmable T5577 tag, I selected a previously saved tag (which was not electra), it could be written without problems.

VladFlorinIlie, 2024-06-20:
I have replicated your experiment here:
https://streamable.com/4a07vk

You can also try with the e-locks T5577 that I have used in the video. You can buy them from EMAG here:
https://www.emag.ro/set-10-bucati-tag-de-proximitate-rfid-e-locks-125-khz-chip-t5577-rewritable-galben-t5577elck/pd/D5RK86MBM/

itrack, 2024-06-20:
Bought, I'll come back with a feedback in 6 days when I arrive :)

My tests so far have been done with:
https://www.aliexpress.com/item/1005005863796435.html?spm=a2g0o.order_list.order_list_main.11.1e331802lOqgr3
and
https://www.a2t.ro/interfoane-videointerfoane/cartela-interfon-electra.html

itrack, 2024-06-22:
Recommended tags arrived faster, same problem, thinking it is a firmware version problem, I reinstalled the original firmware:

[image]

Demo video:
https://www.youtube.com/watch?v=AbJDa9PVRl4

Something's different than what you have @VladFlorinIlie , are you using custom firmware?

itrack, 2024-06-22:
I tried the following cases:

- Read original Electra Tag -> Write original Electra Tag = failed
- Read original Electra Tag -> Write china T5577 = failed
- Read original Electra Tag -> Write e-locks T5577 = failed
- Read e-locks T5577 with Electra Tag clone (written on tag) -> Write china T5577 = failed
- Read e-locks T5577 with Electra Tag clone (written on tag) -> Write another e-locks T5577 = failed
- Read random RFID (non Electra) -> Write e-locks T5577 = success
- Read random RFID (non Electra) -> Write china T5577 = success
- Read random RFID (non Electra) -> Write original Electra Tag = failed

VladFlorinIlie, 2024-06-22:
I do indeed use a custom firmware, Momentum version 004 to be exact. As far as I am aware, there shouldn't be any differences between the original firmware and this custom one when it comes to the Electra RFID protocol.
I guess trying this firmware as well wouldn't hurt :)

itrack, 2024-06-22:
I also installed Momentum v 004 , same result :)
Using the same tags as @VladFlorinIlie , but with a different result, very strange.

Any idea what could be the cause?
Could it be a hardware difference? A debug idea?

dontbug, 2024-06-24:
I have the same problem of not being able to register on the electra chips, but i could on the T5577 chips that i ordered from aliexpress. There were times when i couldn't write from the ordered ones, but it's quite possible that they are not the T5577 model specified by the seller, and i made a refund.

skotopes, 2024-06-24:
@itrack have you tried to read 5577 after write fail?

itrack, 2024-06-24:
@skotopes Yes,  the tag does not change

itrack, 2024-06-24:
Sure:
[electra.zip](https://github.com/user-attachments/files/15956293/electra.zip)

itrack, 2024-06-24:
btw. emulating this tag works

skotopes, 2024-06-24:
> Sure: [electra.zip](https://github.com/user-attachments/files/15956293/electra.zip)

All works:
- Parsing payload
- Emulation
- Writing to 5577
- Reading 5577

itrack, 2024-06-24:
Interesting that writing works for you, how could this be debugged?
I can't explain, I would have thought it was a problem with the RFID module if I couldn't write other cards, but it works, just in this specific case it doesn't.

Are there several hardware versions?

skotopes, 2024-06-24:
> Interesting that writing works for you, how could this be debugged? I can't explain, I would have thought it was a problem with the RFID module if I couldn't write other cards, but it works, just in this specific case it doesn't.
> 
> Are there several hardware versions?

Not in terms of RFID. Also if you can write other types you should be able to write this one too.

itrack, 2024-06-24:
I made another video testing several cases: 

- Trying to write an electra tag (failed)
- Trying to write another tag (successful)
On tags where an electra tag failed, another tag succeeded.

What could be the explanation?

https://www.youtube.com/watch?v=mKwfsRf2ZaI

skotopes, 2024-06-24:
@itrack we don't provide support for 3rd party firmware. If you use it then address all issues to appropriate firmware developers.

itrack, 2024-06-24:
here is a video with the official firmware, it is the same result:
https://www.youtube.com/watch?v=AbJDa9PVRl4

[3 more replies]