flipperdevices/flipperzero-firmware #1505: Saved files are not sorted alphabetically

ppseprus, 2022-07-31:
### Describe the bug.

The files of saved signals are not sorted alphabetically.

SubGHz, RFID and Infra are definitely affecetd.

### Reproduction

Since I could not puzzle out to exact sorting logic, It would be rather hard to write a definitive guide to reproduce the issue.

However, saving multiple files with different names would most probably, eventually, result in the same issue.

skotopes, 2022-09-06:
Never will be. It is a limitation of fatfs driver.

ppseprus, 2022-09-08:
Thanks for the follow up @skotopes !

Willy-JL, 2023-02-12:
> Never will be. It is a limitation of fatfs driver.

https://github.com/ClaraCrazy/Flipper-Xtreme/commit/e98ad6fc2c2bcce58b3b5e726286fb56b2abeef0

and

https://github.com/ClaraCrazy/Flipper-Xtreme/commit/d6dcf0efa26f77f99d9aac9cba00ad6636e1ebc4

:kekw:

Rikarducho, 2023-02-12:
> Never will be. It is a limitation of fatfs driver.

Never say never

skotopes, 2023-02-12:
Try that PR with a couple thousands files

ClaraCrazy, 2023-02-12:
runs fine with ubers repo & the entire irdb

Willy-JL, 2023-02-12:
also kinda your problem at that point? what you even doing with thousands of files in a folder? are you going to scroll through them on flipper? thats gonna take a while mate

skotopes, 2023-02-12:
My point is O(N)

ClaraCrazy, 2023-02-12:
eh, kinda yes. But hes also still got a point, it should ofc run smoothly.. which it does.

skotopes, 2023-02-12:
it will crash flipper on big folders

ClaraCrazy, 2023-02-12:
> it will crash flipper on big folders

It doesnt crash on the IRDB, it doesnt crash on UberGuidoZs entire dump.. how big of a folder are we talking here?

skotopes, 2023-02-12:
Problem with archive that it's running in background while you opening something with it. Any extra memory that you use in archive at that point is going to bite your apps.

We've designed it in a way that it consumes constant amount of ram, what you did totally kills whole idea.

ClaraCrazy, 2023-02-12:
> Problem with archive that it's running in background while you opening something with it. Any extra memory that you use in archive at that point is going to bite your apps.
> 
> We've designed it in a way that it consumes constant amount of ram, what you did totally kills whole idea. 

Now that is info i can work with and understand where you're coming from. Thanks.

skotopes, 2023-02-12:
U r welcome. Also the real solution that you looking for is https://www.codeproject.com/Articles/95721/FAT-32-Sorter embedded into storage service that performs background sorting in FAT, but we decided that it doesn't worth it.

ClaraCrazy, 2023-02-12:
Interesting. Well, thanks for the info. Appreciate it, good work on the FZ btw 🧡 excited to see where this goes