flipperdevices/flipperzero-firmware #1507: Support Noolite (433)

iShift, 2022-07-31:
### Description of the feature you're suggesting.

Add support for recognizing Noolite (433) protocol - it is not encrypted protocol for controlling different smart devices (light/socket/etc) 

So after capturing packages we can show devices in the flipper UI and by clicking - send proper command (enable/disable) 

https://noo.by/produkcia/vykluchateli-besprovodnye.html (in Russian)
https://habr.com/ru/company/contactless/blog/216023/ (In Russian)

https://wirenboard.com/wiki/index.php?title=433MHz/en&mobileaction=toggle_view_desktop  (In English)

Skorpionm, 2022-08-04:
transmission records?

skotopes, 2022-09-06:
@iShift we need raw files with it.

iShift, 2022-09-06:
sorry, can't provide it in near future, if it is critical - we can close the issue @skotopes

closed 2023-03-13 by skotopes

JstSnt, 2025-12-26:
I can provide raw files 
recently found some old switches and use them for holiday lights.