flipperdevices/flipperzero-firmware #1508: Changing the order of Saved Remote Buttons

DrJ735, 2022-07-31:
### Description of the feature you're suggesting.

I saved my first remote the other day and was just thinking about something.  I have my list of buttons and it would be great if I can reorder them.  For an example in the screenshot below I would like to place "Vol_down" under "Vol_Up", and put "Mute" under my volumes. 
[image]

skotopes, 2022-08-01:
We have something like that in archive, there is a reorder option. I'll add this issue to backlog.

DrJ735, 2022-08-01:
Thanks.  It will be just easier to reorder them if you accidently programed the buttons out of order.

gsurkov, 2022-10-14:
I would like to hear some suggestions on how it should be implemented. A mock-up or a text description would be nice.
Can't do this exactly like in the archive, because short and long presses mean different things here.

skotopes, 2024-01-03:
Was implemented some time ago

closed 2024-01-03 by skotopes