flipperdevices/flipperzero-firmware #1512: Toolchain missing clang-format

djsime1, 2022-08-01:
### Describe the bug.

`fbt lint` and `fbt format` fail when clang-format is not installed on the system. As far as I can tell, this is something that should be included in the toolchain as part of it's zero dependencies claim, yet it is not.

### Reproduction

- Checkout firmware to a machine without clang-format installed.
- Run `./fbt lint` and/or `./fbt format`.
- Receive an error stating "clang-format" couldn't be found.

### Target

FBT

### Logs

```Text
$ ./fbt format
python3 scripts/lint.py format applications lib/app-scened-template lib/digital_signal lib/drivers lib/flipper_format lib/infrared lib/nfc_protocols lib/one_wire lib/ST25RFAL002 lib/subghz lib/toolbox lib/u8g2 lib/update_util firmware furi
multiprocessing.pool.RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/workspaces/flipperzero-firmware/toolchain/x86_64-linux/python/lib/python3.9/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/workspaces/flipperzero-firmware/toolchain/x86_64-linux/python/lib/python3.9/multiprocessing/pool.py", line 48, in mapstar
    return list(map(*args))
  File "/workspaces/flipperzero-firmware/scripts/lint.py", line 67, in _format_source
    subprocess.check_call(task)
  File "/workspaces/flipperzero-firmware/toolchain/x86_64-linux/python/lib/python3.9/subprocess.py", line 368, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/workspaces/flipperzero-firmware/toolchain/x86_64-linux/python/lib/python3.9/subprocess.py", line 349, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/workspaces/flipperzero-firmware/toolchain/x86_64-linux/python/lib/python3.9/subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/workspaces/flipperzero-firmware/toolchain/x86_64-linux/python/lib/python3.9/subprocess.py", line 1821, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'clang-format'
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/workspaces/flipperzero-firmware/scripts/lint.py", line 153, in <module>
    Main()()
  File "/workspaces/flipperzero-firmware/scripts/flipper/app.py", line 34, in __call__
    return_code = self.call()
  File "/workspaces/flipperzero-firmware/scripts/flipper/app.py", line 50, in call
    return self.args.func()
  File "/workspaces/flipperzero-firmware/scripts/lint.py", line 144, in format
    if not self._format_sources(sources):
  File "/workspaces/flipperzero-firmware/scripts/lint.py", line 84, in _format_sources
    results = pool.map(self._format_source, tasks)
  File "/workspaces/flipperzero-firmware/toolchain/x86_64-linux/python/lib/python3.9/multiprocessing/pool.py", line 364, in map
    return self._map_async(func, iterable, mapstar, chunksize).get()
  File "/workspaces/flipperzero-firmware/toolchain/x86_64-linux/python/lib/python3.9/multiprocessing/pool.py", line 771, in get
    raise self._value
FileNotFoundError: [Errno 2] No such file or directory: 'clang-format'
scons: *** [phony_format] Error 1
********** ERRORS **********
Failed building phony_format: Error 1
```

### Anything else?

I was working in Github Codespaces when I noticed this oddity. I confirmed it does not occur on a machine with clang-format installed.

skotopes, 2022-08-01:
Hi. That is known issue, we are working on it.

djsime1, 2022-08-30:
I believe this was fixed by https://github.com/flipperdevices/flipperzero-toolchain/commit/258e9c21d0ee9f716dbc830d3fe8a8c769cecc25

closed 2022-08-30 by djsime1