flipperdevices/flipperzero-firmware #1521: Last line of NFC keys dictionaries may be skipped

LoZio, 2022-08-03:
### Describe the bug.

mf_classic_dict.nfc and mf_classic_dict_user.nfc assume there is a newline at the end of the file.
If newline is missing the last key is ignored.
For example in current 0.63.3 there are 1242 keys in the dict, but NFC app says 1241. Last is not used.
I found the problem adding 5 keys to a custom dict and seeing it only says 4 keys. Adding a blank line solved the problem.

### Reproduction

1. Add n keys to mf_classic_dict_user.nfc. Do not add a newline after the last one (save the file after entering the last digit)
2. Upload to Flipper
3. Open NFC app and check the number of keys. It will be n-1
4. Add a newline to the file and upload it again
5. Same as step 3, there will be n keys

skotopes, 2022-09-06:
@gornekich any progress?

Astrrra, 2022-12-08:
This was fixed in #1822. Please open a new issue if the problem persists.

closed 2022-12-08 by Astrrra