flipperdevices/flipperzero-firmware #1522: NFC UID bruteforcing

firefart, 2022-08-03:
### Description of the feature you're suggesting.

on some entry systems that use NFC chips only the UID is checked against a database to check if a chip should be allowed. The problem arises when companys order a large amount of chips, as the ID on them is most likely only incremented by 1. So you can get a valid UID and start counting down or up until you find a valid ID.

I would suggest implemeting this by supplying a start UID and a direction and then either adding or subtracing one from it until cancelled

blacklist-arcc, 2022-08-31:
I think this should be enough: Start UID, increment or decrement value and maybe the delay used to hop to the next id.

skotopes, 2022-09-06:
More like some thing for external application. In theory possible.

closed 2022-09-06 by skotopes