flipperdevices/flipperzero-firmware #1527: Regression on bank card reading

Pooky-s, 2022-08-03:
### Describe the enhancement you're suggesting.

Using firmware 0.62.1 before the NFC application redesign, I could read bank cards (here a VISA debit/credit card) using special actions. 
[image]
The read option using firmware 0.63.3 does not manage to detect bank cards and then reports the same card as a NFC-B. 
[image]
"Special actions" being replaced by "Extra actions", I couldn't find other actions beside "Mf Classic Keys". 
Adding reading options  such as "read bank cards" under "Extra action" or enhancing cards type detection (by testing EMV app presence) would certainly fix the regression.

skotopes, 2022-08-03:
Yep, looks abnormal. @gornekich can you take a look?

SnowLeopard71, 2022-08-03:
I see similar behaviour.   Here is snippet of log from card that read normally followed by one read as NFC-A:

```
47574621 [I][NfcWorker]: ISO14443-4 card detected
47574624 [T][FuriHalNfc]: State change 0 -> 10
47574757 [T][FuriHalNfc]: State change 10 -> 11
47574781 [T][FuriHalNfc]: State change 11 -> 13
47574791 [T][FuriHalNfc]: State change 13 -> 30
47574794 [D][Emv]: Send select PPSE
47574822 [T][Emv]: Select PPSE answer:
```
```
47583122 [I][NfcWorker]: ISO14443-4 card detected
47583128 [T][FuriHalNfc]: State change 0 -> 10
47583256 [T][FuriHalNfc]: State change 10 -> 20
47583426 [T][FuriHalNfc]: Timeout
47583427 [I][NfcWorker]: Unknown card. Save UID
```

skotopes, 2022-09-06:
@gornekich any progress?

hacdias, 2023-06-14:
I've also noticed this recently, any updates? cc @Astrrra

Astrrra, 2023-06-14:
Yes, this feature was completely removed due to people spreading misinformation that it can be used to steal credit card data, which was so severe that no amount of disproving and correcting helped. The feature was a demo with no practical purpose from the beginning, so we decided to cut it, as it brought waaay more harm than good.

hacdias, 2023-06-14:
@Astrrra okay, then I would assume that the [EMV code](https://github.com/flipperdevices/flipperzero-firmware/blob/dev/lib/nfc/protocols/emv.h) would be removed, and this issue can be closed?

skotopes, 2023-06-15:
EMV code will be there for user applications. But yes, we forgot to close this ticket.

closed 2023-06-15 by skotopes