flipperdevices/flipperzero-firmware #1529: Latest Dev Firmware Bricked my Flipper Zero

RyanHuber75214, 2022-08-03:
### Describe the bug.

My Flipper, after the latest update, would not respond to button inputs. I performed a reset, the flipper will now not turn on.

### Reproduction

Update to latest Dev firmware. Perform reset if unresponsive. My model states it is an Atnikora, if that makes a difference. I am using the iPhone version of the App for updates.

RogueMaster, 2022-08-03:
[Try these to put it in DFU](https://github.com/flipperdevices/flipperzero-firmware/blob/dev/documentation/KeyCombo.md)

and reflash using qFlipper

RyanHuber75214, 2022-08-03:
Thank you! I'm alive again, qFlipper fixed it- was a little confused because it didn't seem to boot (no screen activity). 

Cheers!
Ryan

skotopes, 2022-08-04:
Is it reproducible behavior?

RyanHuber75214, 2022-08-04:
Yes, I have just reproduced it by again trying to load the latest Dev Firmware using the iPhone App and then restarting twice as it was unresponsive after the update and unresponsive after the reboot

RyanHuber75214, 2022-08-04:
I did get the screen to come back on after playing with it for a while, but it remained unresponsive. Updating back to Release via qFlipper now.

RyanHuber75214, 2022-08-04:
qFlipper hung once, crashed on the second try, and is finally repairing it...

And, success!

skotopes, 2022-08-04:
Looks like we need your help to reproduce this issue. Do you have debugger or serial adapter?

DrZlo13, 2022-08-04:
Or maybe second Flipper?

RyanHuber75214, 2022-08-04:
I do not have a second Flipper, but I will gladly help if I can. I do have some old serial adapters lying around somewhere- I can try to dig those up. I can also install any dev software you may need, as long as I don't need to purchase licenses

skotopes, 2022-08-04:
Do you have another SD card? If yes, can you format it in flipper and try to update?

If not, then we need your old serial adapter, there is a serial console available on GPIO, baudrate 230400, connect with any terminal emulator you like and get crash log

RyanHuber75214, 2022-08-04:
I have an extra SD card that I have formatted in the Flipper. So, just update now?

RyanHuber75214, 2022-08-04:
I am updating, but I have noticed the Dev version has changed. The one giving me issue yesterday began with a 4, this one begins with a 0. Will update the thread when the firmware update completes.

RyanHuber75214, 2022-08-04:
This one is also causing the same issue. Let me know what are the next steps

skotopes, 2022-08-04:
Next is to take serial adapter and connect to flipper:

11 - flipper GND - adapter GND 
13 - flipper TX - adapter RX

Then using terminal emulator(putty/minicom/etc..) open serial port with 230400 baudrate. And record output when update happens.

[image]

RyanHuber75214, 2022-08-04:
This will take me a little while. Need to catch up on my day-job! Should have it this afternoon

skotopes, 2022-08-04:
Sure, No rush.
Have a nice day.

RyanHuber75214, 2022-08-05:
I think I got rid of my serial devices at my last garage sale, but I have a new one arriving tomorrow. Will update when I can

RyanHuber75214, 2022-08-06:
I got gibberish back:

▒rR▒▒▒%ImEIs='[9E▒▒▒I▒%m#-▒ai▒▒▒rR▒▒▒%ImEIs='m#5
▒▒▒I▒%m#-▒ai▒▒frR▒▒▒%ImEIs='E▒▒▒I▒%q_ma▒ai▒▒▒rR▒▒▒%ImEIs='S5▒▒▒I▒%m#-▒ai▒▒drR▒▒▒%ImEIs='Y-E▒▒▒I▒%m#-▒ai▒▒▒rR▒▒▒%ImEIs='E▒▒▒I▒%-{!#▒ai▒▒g▒rR▒▒▒%ImEIs='E▒▒▒I▒%Y5=▒▒▒I▒%m#-▒ai▒▒gfrR▒▒▒%ImEIs='U;E▒▒▒I▒%m#-▒ai▒▒g▒rR▒▒▒%ImEIs='E▒▒▒I▒%UY{▒ai▒▒g▒rR-'5▒7=5▒▒▒▒rR▒▒▒%ImEIs='_!5▒▒g▒rR▒▒▒%ImEIq=15E▒▒▒I▒%Y)-▒;=5
▒▒▒I▒%m#-▒ai▒▒▒▒rR▒▒▒%ImEIs='S-;▒▒▒I▒%m#-▒ai▒▒▒▒rR▒▒▒%ImEIs='Y;q/
                                                                 E▒▒▒I▒%m#-▒ai▒▒gegrR▒▒▒%ImEIs='c39E▒▒▒I▒%m#-▒ai▒▒gefrR▒▒▒%ImEIs='y!%E▒▒▒I▒%m#-▒ai▒▒ge▒rR▒▒▒%ImEIs='E▒▒▒I▒%s}WsY▒ai▒▒geerR▒▒▒%ImEIs'-5
▒▒▒I▒%▒▒▒s-▒5ퟣ▒▒▒▒▒▒▒{-'7▒7=5▒▒ퟙ▒▒▒▒▒▒▒▒▒▒▒q-▒y!%%-▒▒▒9▒▒▒33▒▒▒▒▒▒▒▒▒▒▒q-▒{#9/▒▒ퟣ▒▒▒▒▒▒gedrR▒▒▒%ImEIs'-5
▒▒▒I▒%=▒55▒▒gecrR▒▒▒%ImEIs'-5
▒▒▒I▒%=▒55▒[9Y▒g▒▒rR▒▒▒%ImEIy!▒E▒▒▒I▒%y!▒▒=7▒▒g▒▒rR▒▒▒%ImEIs'-5
▒▒▒I▒%=▒55▒{Y▒g▒▒rR▒▒▒%ImEIs'-5
▒▒▒I▒%=▒55▒y'-Y▒g▒▒rR▒▒▒%ImEIY=57YE▒▒▒I▒%g!=7-#1▒▒▒-#▒▒;▒5-#1▒▒▒gd▒rR▒▒▒%ImEIs'-5
▒▒▒I▒%=▒55▒w-='!1Y▒gd▒rR▒▒▒%ImEIs'-5
▒▒▒I▒%=▒55▒w!'/-#Y▒gd▒rR▒▒▒%ImEIs'-5
▒▒▒I▒%=▒55▒w5)!Y▒gddrR▒▒▒%IQEIy'-Y
▒▒▒I▒%Y)-57▒ygm▒5-!#▒!5#▒▒75-95▒-#▒59-='▒=%!75▒▒gdcrR▒▒▒%ImEIs'-5
▒▒▒I▒%=▒55▒q-Y▒g▒▒rR▒▒▒%ImEIY=57YE▒▒▒I▒%g!=7-#1▒▒▒-#▒▒7!'/-#▒=5▒▒▒g,▒rR▒▒▒%ImEIs'-5
▒▒▒I▒%=▒55▒m#Y▒g▒▒rR▒▒▒%ImEIs'-5
▒▒▒I▒%=▒55▒g!=75▒g,▒rR▒▒▒%ImEIs'-5
▒▒▒I▒%=▒55▒c!-3-9=-!#Y▒g▒▒rR▒▒▒%ImEIg!=75
5%▒=/!!)▒▒gcgrR▒▒▒%ImEIs'-5
▒▒▒I▒%=▒55▒_!5▒gc▒rR▒▒▒%ImEIs'-5
▒▒▒I▒%=▒55▒Y!15Y▒gccrR▒▒▒%ImEIs'-5
▒▒▒I▒%55▒=9!%'55▒▒▒▒▒rR▒▒▒%ImEIY!15m#E▒▒▒I▒%Y=▒▒▒▒▒rR▒▒▒%ImEIY!15m#E▒▒▒I▒%y!#3-1▒▒=▒▒▒;▒▒▒▒▒▒=7▒▒▒▒▒▒▒▒=15▒-
9'5▒▒▒▒▒▒▒▒▒▒V▒rR▒▒▒%ImEIY!15m#E▒▒▒I▒%e!#57▒▒▒6▒rR▒▒▒%ImEIY!15uE▒▒▒I▒%9=▒755957▒▒d▒▒rR▒▒▒%ImEIY!15uE▒▒▒I▒%9=▒%!#57▒▒d6▒rR▒▒▒%ImEIc!-3-9=-!#Y
▒▒▒I▒%'!=7-#1▒5-#1▒3▒▒-#▒▒#!-3-9=-!#▒5-#1▒▒▒dggrR▒▒▒%ImEIg!=75
▒▒▒I▒%Y=▒▒d▒▒rR▒▒▒%ImEIg!=75
▒▒▒I▒%{-'7-#1▒%=-#▒%5#▒▒d▒▒rR▒▒▒%ImEIg!=75
▒▒▒I▒%{-'7-#1▒'1-#▒%5#▒▒d▒▒rR▒▒▒%ImEIg!=75
▒▒▒I▒%{-'7-#1▒75;1▒%5#▒▒df▒rR▒▒▒%ImEIg!=75
▒▒▒I▒%{-'7-#1▒5-#1▒%5#▒▒dffrR▒▒▒%ImEIg!=75
▒▒▒I▒%Y=7▒▒df▒rR▒▒▒%ImEIg!=75
▒▒▒I▒%Y=▒▒U7=5▒▒dY▒rR▒▒▒%ImEIc!-3-9=-!#Y
▒▒▒I▒%'!=7▒995▒▒dd▒rR▒▒▒%IQEI{Y
▒▒▒I▒%Y)-57▒{W▒-#-▒▒75-95▒-#▒59-='▒=%!75▒▒dderR▒▒▒%ImEIy!▒E▒▒▒I▒%y▒▒;!!▒9!%'557▒▒%!75▒▒Y=9)▒▒dddrR▒▒▒%ImEIy!▒E▒▒▒I▒%y!▒▒▒sUY▒▒▒▒▒▒▒▒▒%5%▒▒▒▒▒▒▒3'=/▒▒▒=15▒▒d▒▒rR5▒▒▒▒3'=/▒▒▒▒=15▒▒gf,▒rR▒▒▒%ImEI}#-%=-!#e=#=15
▒▒▒I▒%Y5'59▒▒g▒Ag=!A▒▒▒▒▒▒▒=#-%=-!#▒▒gfc▒rR▒▒▒%ImEI}#-%=-!#e=#=15
▒▒▒I▒%U#'!=7▒=#-%=-!#▒▒g▒Ag=!A▒▒▒▒▒▒▒▒gٶ▒rR▒▒▒%ImEIY=57YE▒▒▒I▒%g!=7-#1▒▒▒-#▒▒75)!▒5-#1▒▒▒gYֿrR▒▒▒%ImEIW=/E▒▒▒I▒%{=9)-#1▒▒▒▒-#▒▒▒▒▒▒▒gKI▒▒▒▒%ImEIs='[9E▒▒I▒%h▒-▒aiP▒fKI▒▒▒▒%ImEIs='m#5
▒▒I▒%h▒-▒aiP▒eKI▒▒▒▒%ImEIs='E▒▒I▒%▒}▒▒▒aiP▒dKI▒▒▒▒%ImEIs='S5▒▒I▒%h▒-▒aiP▒cKI▒▒▒▒%ImEIs='Y-E▒▒I▒%h▒-▒aiP▒g▒KI▒▒▒▒%ImEIs='E▒▒I▒%▒▒U!#▒aiP▒g▒KI▒▒▒▒%ImEIs='E▒▒I▒%▒▒-'5▒7=5P▒▒▒KI▒▒▒▒%ImEIs='_!5'm▒yE▒▒I▒%h▒-▒aiP▒g▒KI▒▒▒▒%ImEIq=15E▒▒I▒%▒▒j▒;=5
▒▒I▒%h▒-▒ai▒▒▒▒KI▒▒▒▒%ImEIs='y!%E▒▒I▒%[▒▒▒
                                          ▒A▒▒▒KI▒▒▒▒%ImEIs='E▒▒I▒%0▒]▒e▒aiP▒▒▒KI▒▒▒▒%ImEIs'-5
▒▒I▒%y▒o▒▒5▒#)#!#▒▒[}e▒▒▒▒{-'7▒7=5▒▒ퟕ▒▒▒▒▒▒▒▒▒▒▒q-▒y!%%-▒▒▒7▒▒▒▒▒▒7▒▒▒▒▒▒▒▒▒▒q-▒{#9/▒▒▒75y=f▒KI▒▒▒▒%ImEIs'-5
▒▒I▒%FQ▒-#1▒55▒▒f▒KI▒▒▒▒%ImEIs'-5
▒▒I▒%FQ▒-#1▒55▒q-Y▒feKI▒▒▒▒%ImEIy!▒E▒▒I▒%^H▒▒▒▒=7y=fdKI▒▒▒▒%ImEIs'-5
▒▒I▒%▒t▒▒-#1▒55▒m#Y▒▒▒KI▒▒▒▒%ImEIs'-5
▒▒I▒%FQ▒-#1▒55▒c!-3-9=-!#Y=▒▒KI▒▒▒▒%ImEIs'-5
▒▒I▒%FQ▒-#1▒55▒Y!15Y=e▒KI▒▒▒▒%ImEIs'-5
▒▒I▒%FQ▒-#1▒55▒U7=5=e▒KI▒▒▒▒%ImEIY!15uE▒▒I▒%NO▒ӻ▒▒5957P▒e▒KI▒▒▒▒%ImEIs'-5
▒▒I▒%FM▒▒)▒ʿ=9!%'55y=eY▒KI▒▒▒▒%ImEIY!15uE▒▒I▒%▒▒▒7▒%!#57P▒ee▒KI▒▒▒▒%ImEIc!-3-9=-!#Y
▒▒I▒%R▒j▒▒▒5-#1▒3▒▒-#▒▒#!-3-9=-!#▒5-#1▒y=ee▒KI▒▒▒▒%IuEIc!-3-9=-!#Y
▒▒I▒%R▒▒▒j:▒▒:▒k▒='-7▒#=%5▒=/y=e▒▒KI▒▒▒▒%ImEIc!-3-9=-!#Y
▒▒I▒%FO▒i▒▒▒5-#1▒!▒▒▒-#▒▒#!-3-9=-!#▒5-#1▒▒▒e▒▒KI▒▒▒▒%IuEIc!-3-9=-!#Y
▒▒I▒%FO▒ӛ▒j:▒▒:▒k▒='-7▒#=%5▒=/▒▒cgdKI▒▒▒▒%ImEIy!▒E▒▒I▒%^▒▒

                                                          ▒▒9!%'557▒▒%!75▒▒Y=9)▒▒cgcKI▒▒▒▒%ImEIy!▒E▒▒I▒%^H▒Ӷ▒sUY▒▒▒▒▒▒▒▒▒%5%▒▒▒▒▒▒▒3'=/▒▒▒=15▒▒c▒▒KI▒▒▒▒%ImEIy!▒5▒▒▒▒3'=/▒▒▒▒=15▒▒c▒▒KI▒▒▒▒%IQEIU7Q!eE▒▒I▒%▒u▒▒J▒5-▒▒7=5▒▒c▒▒KI▒▒▒▒%ImEIU7Q!eE▒▒I▒%X▒+▒]▒-▒▒▒▒▒='▒▒w▒s▒▒}}▒▒▒▒▒w▒s▒▒}}▒▒5▒▒y=cf▒KI▒▒▒▒%ImEIU7Q!eE▒▒I▒%X▒+▒]▒-▒▒▒▒▒='▒▒▒▒▒▒▒ss▒▒▒▒▒▒▒▒▒▒ss▒▒5▒▒y=c▒▒KI▒▒▒▒%ImEIU7Q!eE▒▒I▒%X▒+▒]▒-▒▒▒▒▒='▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒5▒▒y=cY▒KI▒▒▒▒%ImEIU7Q!eE▒▒I▒%X▒+▒]▒-▒▒▒▒▒='▒▒▒▒▒▒▒ss▒▒▒▒▒▒▒▒▒▒ss▒▒5▒▒▒▒cY▒KI▒▒▒▒%ImEIU7Q!eE▒▒I▒%X▒+▒]▒-▒▒▒▒▒='▒▒▒▒▒▒▒ss▒▒▒▒▒▒▒▒▒▒ss▒▒5▒▒▒▒cY▒KI▒▒▒▒%ImEIU7Q!eE▒▒I▒%X▒+▒]▒-▒▒▒▒▒='▒▒▒▒▒▒▒ss▒▒▒▒▒▒▒▒▒▒ss▒▒5▒▒▒▒cegKI▒▒▒▒%ImEIU7Q!eE▒▒I▒%X▒+▒]▒-▒▒▒▒▒='▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒5▒▒▒▒ce▒KI▒▒▒▒%ImEIU7Q!eE▒▒I▒%X▒+▒]▒-▒▒▒▒▒='▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒5▒▒▒▒ce▒KI▒▒▒▒%ImEIU7Q!eE▒▒I▒%X▒+▒]▒-▒▒▒▒▒='▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒5▒▒▒▒cecKI▒▒▒▒%ImEIU7Q!eE▒▒I▒%X▒+▒]▒-▒▒▒▒▒='▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒5▒▒▒▒c▒▒KI▒▒▒▒%ImEIU7Q!eE▒▒I▒%X▒+▒]▒-▒▒▒▒▒▒='▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒5▒▒▒▒c▒▒KI▒▒▒▒%ImEIU7Q!eE▒▒I▒%X▒+▒]▒-▒▒▒▒▒▒='▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒5▒▒▒▒c▒▒KI▒▒▒▒%ImEIU7Q!eE▒▒I▒%X▒+▒]▒-▒▒▒▒▒▒='▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒5▒▒▒▒c▒▒KI▒▒▒▒%ImEIU7Q!eE▒▒I▒%X▒+▒]▒-▒▒▒▒▒▒='▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒5▒▒▒▒cdgKI▒▒▒▒%ImEIU7Q!eE▒▒I▒%X▒+▒]▒-▒▒▒▒▒▒='▒▒▒▒▒▒▒w▒▒▒▒▒▒▒▒▒▒▒w▒▒▒5▒▒▒▒cdfKI▒▒▒▒%ImEIU7Q!eE▒▒I▒%X▒+▒]▒-▒▒▒▒▒▒='▒▒}▒{▒y▒▒▒▒▒▒▒}▒{▒y▒▒▒▒5▒▒▒▒▒KI▒▒▒▒%ImEIs='[9E▒▒I▒%h▒-▒aiP▒▒KI▒▒▒▒%ImEIs='m#5
▒▒I▒%h▒-▒ai▒▒▒KI▒▒▒▒%ImEIs='E▒▒I▒%▒}▒▒▒aiP▒eKI▒▒▒▒%ImEIs='S5▒▒I▒%h▒-▒ai▒▒▒KI▒▒▒▒%ImEIs='Y-E▒▒I▒%h▒-▒aiP▒cKI▒▒▒▒%ImEIs='E▒▒I▒%▒▒U!#▒aiP▒g▒KI▒▒▒▒%ImEIs='E▒▒I▒%▒▒T▒▒I▒%h▒-▒ai▒▒gfKI▒▒▒▒%ImEIs='U;E▒▒I▒%▒kj▒▒aiP▒geKI▒▒▒▒%ImEIs='E▒▒I▒%▒e▒▒aiP▒g▒KI-'5▒7=5▒▒▒▒KI▒▒▒▒%ImEIs='_!5▒gcKI▒▒▒▒%ImEIq=15E▒▒I▒%@▒j▒;=5
▒▒I▒%h▒-▒ai▒▒▒▒KI▒▒▒▒%ImEIs='S-;▒▒I▒%h▒-▒ai▒▒▒▒KI▒▒▒▒%ImEIs='Y;q/
                                                                 E▒▒I▒%h▒-▒ai▒▒gegKI▒▒▒▒%ImEIs='c39E▒▒I▒%Ѝ-▒aiP▒ge▒KI▒▒▒▒%ImEIs='y!%E▒▒I▒%[▒▒▒
[... cut ...] ▒▒▒▒{{▒▒▒o▒▒▒▒▒▒▒▒▒▒▒▒▒{[o{{▒▒▒o▒{▒▒▒▒▒{▒▒▒o▒▒{▒▒▒▒▒▒{{▒{▒{{▒{{▒o{▒▒▒▒▒▒▒{o▒o{{{{{▒{▒▒{{▒▒k▒{▒▒▒{▒▒{▒k{▒▒o{{o{▒{▒▒▒▒▒▒o{▒▒▒▒▒▒▒▒{▒▒{▒▒▒{▒▒{▒▒{{▒▒▒o▒{▒▒o▒{{▒▒{▒{▒▒▒▒▒▒▒▒▒▒{▒{{▒▒{▒{{{{{▒▒o▒▒▒▒▒▒{{▒{{o▒{{{{{{▒{▒▒▒▒▒▒▒▒{{▒oo▒▒▒▒{▒▒▒▒▒o{{{{{▒▒{{{▒{{▒▒{▒▒{▒▒{▒▒o▒▒▒▒{{▒▒{▒▒{▒{o{o{{▒▒{▒▒▒{▒▒▒▒{▒{{k▒▒▒{▒{▒{▒{{▒▒▒{▒{{▒▒{{▒{{▒{▒{▒▒▒▒▒▒{o{▒▒▒{{k{▒▒▒{{oo▒▒o{{{{▒▒{▒▒▒o{▒{{{{{{{▒{▒▒o▒k{▒{o{oo{{ko{{k{{▒{▒▒▒▒▒▒{▒▒{k{{{{{▒{{▒▒{▒{{{{{{{▒{▒▒o▒{{▒▒{{▒{▒▒{{{{{o{o{{▒{{▒o{o{▒{{{k▒o{▒▒▒▒▒{▒▒{▒{▒▒{▒▒{{{▒▒o▒{▒{{▒{{▒▒{{▒{{

skotopes, 2022-08-07:
what settings you were using ?

RyanHuber75214, 2022-08-07:
Okay, first off, using MobaXTerm as my terminal emulator. Using my USB/Serial connector on COM5. Settings are:
PuTTY Serial Engine
Data bits: 8
Stop bits: 1
Parity: None
Flow Control: Xon/Xoff

sodoku, 2022-08-08:
Did you set the baudrate to 230400?

RyanHuber75214, 2022-08-08:
Yes, I am using a baud rate of 230,400

RyanHuber75214, 2022-08-12:
Well, I'm not sure what all changes have been made, but the latest Dev version no longer causes issues. I took a break for a few days, but everything seems resolved for now!

closed 2022-09-06 by skotopes