flipperdevices/flipperzero-firmware #1530: NFC: improve dict attack by trying the last successful key before starting over on next sector

u1735067, 2022-08-04:
### Describe the enhancement you're suggesting.

Some NFC sectors may share the same NFC key, so in order to reduce dictionary attack time, a simple optimization may be to try the last successful key before starting the full dictionary attack. If it's not the same key, you lose 1 attempt delay, but if it's the same key, you gain `position in the list` attempts delay.

Another optimization would be to try all previously successful keys for this tag, not just the last one.

skotopes, 2022-09-06:
Please check latest firmware and reopen issue if it persists.

u1735067, 2022-09-06:
Can you reopen? It's not implemented, tested with a tag having the same sector key, each sector is going through the full dictionary instead of reusing the previous found key(s) as suggested:
[g68SJ4B8WUPkeMH.webm](https://user-images.githubusercontent.com/1735067/188739589-1b39879e-cc87-4294-8274-86b312768a24.webm)

skotopes, 2022-09-07:
Sure. @gornekich Can you take a look?

gornekich, 2022-09-07:
We have PR #1488 from community, it is stil WIP. I think I will do it myself, if it is not ready in a few days

shelld0n, 2022-09-19:
@gornekich , do you know if this PR is still going ?

gornekich, 2022-09-19:
We are working on it in our team.

gornekich, 2022-09-19:
In a different branch

AkechiShiro, 2022-10-18:
I believe this issue is fixed by #1885, no ?

u1735067, 2022-10-22:
Seems to be working for me, thank you :)

closed 2022-10-22 by u1735067