flipperdevices/flipperzero-firmware #1534: Hitag2 Support

eben80, 2022-08-04:
### Description of the feature you're suggesting.

Would it be possible to support Hitag2 tags under the 125kHz RFID function?

I'm not sure if the [code](https://github.com/Proxmark/proxmark3/blob/master/armsrc/hitag2.c) in this project would be of any help.

csBlueChip, 2022-08-20:
If you are referring to cracking HiTag2 - it is computationally very expensive to crack - and certainly not in the (realistic) remit of a FZ.
Kev Sheldrake did a talk on this at 44con 2017 https://www.youtube.com/watch?v=abx1hQDCKyg&ab_channel=44CONInformationSecurityConference

However, I don't see why you couldn't use a FZ to collect the data for offline attacks.

COssfeld, 2022-09-06:
@csBlueChip I don't know if HITAG 2 uses the same system as HITAG S but for me it would be enough if flipper would support reading the uid and other data in plain mode without auth and also emulate the uid. If I am rememeber correctly you need an extra IC for auth.
https://www.nxp.com/docs/en/data-sheet/HTSICH56_48_SDS.pdf

daandewitte, 2023-05-02:
fyi, I recently finished my developments for an external app which can read & emulate the HITAG 1 cards.
It is supporting both basic & advanced mode, though indeed only plain mode. I could add support for encrypted mode (assuming the user knows the key), but since I can't find any documentation on the encryption method used I cannot implement it.

Since all hitag types are quite similar in nature (meaning they all work with bidirectional cmd & reply modes, vs most other onedirectional LFRFID tags) I believe I could also add support for HITAG 2 and S.
Tricky thing is that I don't have any HITAG 2 or S cards nor readers to test with...

gsp8181, 2023-05-03:
@daandewitte 

I have some hitag2 cards so if you send me the app I can have a crack with it and test it against hitag2

daandewitte, 2023-05-05:
fyi @skotopes @Astrrra @DrZlo13 
i'm working on including the hitag support into the main rfid app as part of the existing read & emulate routines.

testing out on my dev first, so you'll likely be seeing a pull request in few days from now.

sekwah41, 2024-05-29:
Any news on this?

stuiterveer, 2024-06-13:
[For as far as I understood](https://forum.flipper.net/t/flipper-as-hitag-2-transponder-programmer/7507/5) the cracking/challenge response is a tough subject, but (from an outsider's perspective) it does look like normal password encoding is cleartext and you "just" need to know the password (and sniffing it is relatively easy as well since it's sent cleartext).

So it does look like, even if there are limitations to rolling out Hitag2 support, there's some great progress that's relatively easy to implement. @blackvault88 while I lack the knowledge to help out with the code itself, I can offer help in terms of testing and debugging of the protocol, or even sourcing hardware that might help you. Feel free to reach out to me by email if I can help out. I'd love to see more support for Hitag in general!