flipperdevices/flipperzero-firmware #1536: Stuck in remote when using Back button while saving

tomini, 2022-08-04:
### Describe the bug.

I was adding buttons to a remote, it saved successfully, but I got stuck in **Saved screen**  loop.

**Expected behavior**
Well, go back to the remote screen and be able to go back to desktop/favorites and continue using the Flipper

**Actual behavior**
Got stuck on the screen I went back to and had to reboot the Flipper.

**Additional troubleshooting**
1. Does the issue persist if you try multiple times? **Yes.**
2. Does reconnecting/rebooting Flipper solve it? **Yes**
3. Does deleting the button solve the issue? **No**
4. Does going through Edit options solve the issue? **No**

### Reproduction

1. Open remote from favorite (pinned)
2. Add new button
3. Read the IR signal
4. Save (name doesn't matter)
5. Click `BACK` when you see **Saved** screen with Flipper
6. Now you are back on the remote screen, button is saved, but you can't exit this screen
[Video](https://youtu.be/aMkXJj1Bf7s)

### Target

0.63.3

### Logs

```Text
Sorry, I tried my best. Turned Debug Mode ON, changed Log Level to Debug and the debug.pcap was either impossible to copy to PC (0 b size in the end) or it wasn't even there.
Tried it several times.
It's really easy to reproduce, so at least you can try to salvage the debug logs from yours.
```

### Anything else?

This seems to be a minor bug. As you have to do it from favorites and then click Back to save yourself like one second. But it's still a bug that has to be reported.

gsurkov, 2022-08-04:
Confirmed, fix is in progress.

tomini, 2022-08-04:
> Confirmed, fix is in progress.

And did I mess up with the logs or did the bug or reboot cause it to be broken?

gsurkov, 2022-08-04:
There is a logical error in the Infrared application that leads to the incorrect screen being shown. It happens only when opening a remote from favourites of file browser.
You have provided enough information so that the bug could be easily reproduced and confirmed, which I am thankful for.

bbock, 2022-08-04:
Is this a duplicate of #1426 ?

gsurkov, 2022-08-04:
Looks like it! Anyway, it should be now fixed here #1538 (not in the main branch yet)

closed 2022-08-08 by gsurkov