flipperdevices/flipperzero-firmware #1544: Bluetooth Remote plugin stops working after restart.

ClusterM, 2022-08-05:
### Describe the bug.

Bluetooth remote plugin works fine after pairing with device. But this plugin stops working after restart. Bluetooth device (smartphone/PC) still connected to Flipper but there is no response when pressing virtual buttons.

The plugin continues to work if pairing is repeated.

Tested on Windows 10 and Android 12.

### Reproduction

1. Start "Bluetooth Remote" plugin.
2. Pair phone/PC with Flipper.
3. Make sure that you can control your device via Flipper.
4. Close plugin.
5. Start "Bluetooth Remote" plugin again.
6. Try to control your device via Flipper. It's not working anymore.

### Logs

```Text
3408099 [I][SavedStruct]: Loading "/int/.desktop.settings"
3408109 [I][LoaderSrv]: Starting: Bluetooth Remote
3408110 [I][AnimationManager]: Unload animation 'L1_Read_books_128x64'
3408150 [I][SavedStruct]: Loading "/int/.bt.settings"
3408163 [I][FuriHalBt]: Disconnect and stop advertising
3408165 [I][BtGap]: Stop advertising
3408167 [D][BtGap]: set_non_discoverable success
3408169 [I][FuriHalBt]: Stop current profile services
3408177 [I][FuriHalBt]: Stop BLE related RTOS threads
3408200 [I][FuriHalBt]: Reset SHCI
3408313 [I][FuriHalBt]: Start BT initialization
3408316 [I][Core2]: Core2 started
3408319 [I][Core2]: C2 boot completed, mode: Stack
3408321 [I][Core2]: Core2: FUS: 1.2.0, mem 16/0, flash 6 pages
3408322 [I][Core2]: Core2: Stack: 1.13.3, branch 0, reltype 2, stacktype 3, flash 30 pages
3408326 [I][Core2]: Radio stack started
3408328 [I][Core2]: Flash activity control switched to SEM7
3408345 [D][BtBatterySvc]: Updating power state characteristic
3408354 [I][BtSrv]: Bt App started
3408356 [I][BtGap]: Start advertising
3409989 [D][ViewDispatcher]: View changed while key press 0x200100c0 -> 0x20011790. Sending key: OK, type: Release, sequence: 0x23c to previous view port
3414107 [I][BtGap]: Connection parameters: Connection Interval: 35 (43 ms), Slave Latency: 0, Supervision Timeout: 500
3414271 [D][BtGap]: Slave security initiated
3414272 [D][BtGap]: Bond lost event. Start rebonding
3414486 [I][BtGap]: Verify numeric comparison: 589312
3419868 [I][BtSrv]: Changed addr start: 20030BB8, size changed: 4
3419870 [I][BtSrv]: Changed addr start: 20030BB8, size changed: 88
3419872 [I][BtGap]: Pairing complete
3419873 [I][SavedStruct]: Saving "/int/.bt.keys"
3419882 [D][StorageInt]: Device erase: page 2, translated page: c3
3419956 [D][StorageInt]: Device sync: skipping, cause
3419959 [I][SavedStruct]: Saving "/int/.bt.keys"
3419967 [D][StorageInt]: Device erase: page 3, translated page: c4
3420062 [D][StorageInt]: Device sync: skipping, cause
3420065 [D][BtBatterySvc]: Updating battery level characteristic
3420597 [I][BtGap]: Connection parameters event complete
3420598 [I][BtGap]: Connection parameters: Connection Interval: 6 (7 ms), Slave Latency: 0, Supervision Timeout: 500
3420600 [W][BtGap]: Unsupported connection interval. Request connection parameters update
3420663 [D][BtGap]: Connection parameters accepted
3421569 [I][BtGap]: Connection parameters event complete
3421570 [I][BtGap]: Connection parameters: Connection Interval: 36 (45 ms), Slave Latency: 0, Supervision Timeout: 500
3430298 [D][ViewDispatcher]: View changed while key press 0x20011790 -> 0x200100c0. Sending key: Back, type: Release, sequence: 0x24c to previous view port
3430751 [D][ViewDispatcher]: View changed while key press 0x200100c0 -> 0. Sending key: Back, type: Release, sequence: 0x24d to previous view port
3430756 [I][SavedStruct]: Loading "/int/.bt.settings"
3430778 [I][FuriHalBt]: Disconnect and stop advertising
3430780 [I][BtGap]: Stop advertising
3430784 [D][BtGap]: terminate success
3430787 [E][BtGap]: set_non_discoverable failed 12
3430790 [I][FuriHalBt]: Stop current profile services
3430807 [I][FuriHalBt]: Stop BLE related RTOS threads
3430830 [I][FuriHalBt]: Reset SHCI
3430943 [I][FuriHalBt]: Start BT initialization
3430946 [I][Core2]: Core2 started
3430949 [I][Core2]: C2 boot completed, mode: Stack
3430951 [I][Core2]: Core2: FUS: 1.2.0, mem 16/0, flash 6 pages
3430952 [I][Core2]: Core2: Stack: 1.13.3, branch 0, reltype 2, stacktype 3, flash 30 pages
3430956 [I][Core2]: Radio stack started
3430958 [I][Core2]: Flash activity control switched to SEM7
3430976 [D][BtBatterySvc]: Updating power state characteristic
3430980 [I][BtSrv]: Bt App started
3430981 [I][BtGap]: Start advertising
3431085 [I][LoaderSrv]: Application thread stopped. Free heap: 119336. Thread allocation balance: 0.
3431343 [I][AnimationManager]: Select 'L1_Read_books_128x64' animation
3431347 [I][AnimationManager]: Load animation 'L1_Read_books_128x64'
3431374 [I][SavedStruct]: Loading "/int/.desktop.settings"
3433939 [I][SavedStruct]: Loading "/int/.desktop.settings"
3433949 [I][LoaderSrv]: Starting: Bluetooth Remote
3433951 [I][AnimationManager]: Unload animation 'L1_Read_books_128x64'
3433990 [I][SavedStruct]: Loading "/int/.bt.settings"
3434004 [I][FuriHalBt]: Disconnect and stop advertising
3434005 [I][BtGap]: Stop advertising
3434008 [D][BtGap]: set_non_discoverable success
3434009 [I][FuriHalBt]: Stop current profile services
3434017 [I][FuriHalBt]: Stop BLE related RTOS threads
3434041 [I][FuriHalBt]: Reset SHCI
3434154 [I][FuriHalBt]: Start BT initialization
3434157 [I][Core2]: Core2 started
3434160 [I][Core2]: C2 boot completed, mode: Stack
3434162 [I][Core2]: Core2: FUS: 1.2.0, mem 16/0, flash 6 pages
3434163 [I][Core2]: Core2: Stack: 1.13.3, branch 0, reltype 2, stacktype 3, flash 30 pages
3434167 [I][Core2]: Radio stack started
3434169 [I][Core2]: Flash activity control switched to SEM7
3434187 [D][BtBatterySvc]: Updating power state characteristic
3434195 [I][BtSrv]: Bt App started
3434197 [I][BtGap]: Start advertising
3435744 [I][BtGap]: Connection parameters: Connection Interval: 35 (43 ms), Slave Latency: 0, Supervision Timeout: 500
3435811 [D][ViewDispatcher]: View changed while key press 0x20010078 -> 0x20011750. Sending key: OK, type: Release, sequence: 0x24f to previous view port
3435893 [D][BtGap]: Slave security initiated
3436153 [I][BtGap]: Pairing complete
3436155 [D][BtBatterySvc]: Updating battery level characteristic
3451086 [D][ViewDispatcher]: View changed while key press 0x20011750 -> 0x20010078. Sending key: Back, type: Release, sequence: 0x26e to previous view port
3452373 [D][ViewDispatcher]: View changed while key press 0x20010078 -> 0x20012830. Sending key: OK, type: Release, sequence: 0x272 to previous view port
```

### Anything else?

I'm using firmware compiled from 040558c8f989aa36a50f86276f847dfb73cfa8e2
From time to time it still works. I can't understand what I need to do to make it work.
Also, sometimes this plugin is not reconnecting to smartphone automatically.

ClusterM, 2022-08-05:
Actually it's not working anymore after https://github.com/flipperdevices/flipperzero-firmware/blob/d147190d6173f427bbedce988cb28eff22ca692e/firmware/targets/f7/furi_hal/furi_hal_bt.c#L266
call.

ClusterM, 2022-08-08:
Please check it, i really want to control music on my phone using Flipper. Phones has no multimedia buttons now.

ClusterM, 2022-08-10:
Sometimes only keyboard works but not mouse and media buttons.

gravityfox, 2022-08-25:
I am also having this issue and would be interested in a fix.

coded-with-claws, 2022-09-04:
I also have this issue : after successfully pairing with Kodi on raspberry pi, the Keynote allows to navigate into Kodi. But if I quit the bluetooth plugins, and re-enter bluetooth plugins, the LED is blue but Keynote no longer works.

skotopes, 2022-09-06:
@gornekich can you check?

gornekich, 2022-09-07:
Reproduced, planed to fix

coded-with-claws, 2022-09-09:
Just FYI, I no longer have the problem on 0.66 with my raspi Kodi. I can navigate between IR remote and BT keynote plugin without any problem.
I wonder if anything was done about that in 0.66 ?

skotopes, 2023-03-13:
there were couple fixes, do you still get this problem?

afaonline, 2023-03-14:
Firmware 0.79.1, problem is fixed! I had to forget the link and pair again on smartphone, and now it works perfectly. Thank you!

skotopes, 2023-03-14:
Nice )

closed 2023-03-14 by skotopes

ClusterM, 2023-03-31:
Yes, works fine for me too, thank you!