flipperdevices/flipperzero-firmware #1547: Frequency analyzer history and capture with GPS

ResistanceIsUseless, 2022-08-05:
### Description of the feature you're suggesting.

- In Frequency analyzer add the following options:
1.  Ability to change sensitivity
2. Show last X amount of captures
3. Ability to log the signal to SD with timestamp
4. Add GPS location (GPS over GPIO's)

Also, I'm not sure if the STM32 and CC1101 could handle this, but having the option to switch from Frequency analyzer to raw capture automatically and continually log the time and GPS location would be nice. Basically have it scan for frequencies and attempt to capture anything it sees and save the capture if it gets anything.

If there are hardware limitations to this, maybe it could be done by connecting another CC1101 over GPIO or Flipper > ESP32 module > Additional CC1101. Where the ESP32 handles captures with the CC1101. There would probably need to be some type of configuration to make sure the additional CC1101 doesn't have frequency shift or whatever it's called.

### Anything else?

I wouldn't mind trying to help code this. I'm just experienced with embedded development. If there is any documentation to help me get started i wouldn't mind giving it a shot.

skotopes, 2022-09-06:
Good idea for external application

closed 2022-09-06 by skotopes

Stichoza, 2025-07-14:
I made a webapp related to this – **[flipper-map.stichoza.com](https://flipper-map.stichoza.com/)** – you can visualize captures on map (connected using USB).

Currently you have to add GPS coordinates manually in .sub files, but it would be awesome if coordinates could be added automatically using GPS module.

Flipper Map looks like this in case you have coordinates included in files:

[image]