flipperdevices/flipperzero-firmware #1549: Error "Can not parse file" on a NFC tag

coded-with-claws, 2022-08-06:
### Describe the bug.

After using the NFC app to read and save a NFC-A tag, it won't allow to open it : "Can not parse file".
It happens on version 0.63.3. It worked in previous version 0.62 with the same NFC tag.

I attach the file, which I edited to replace the characters of UID with "X" characters.

### Reproduction

1. Open menu, NFC, Read with a NFC tag under the flipper.
2. It displays "NFC-A" with ATQA, SAK and UID information.
3. Click on Save, type the name of the file.
4. Open menu, NFC, Saved, choose the file previously saved.
5. It displays "Can not parse file".

gornekich, 2022-08-07:
Thanks!
Now flipper doesn't support 8-byte UIDs. Will try to add these cards soon

coded-with-claws, 2022-08-31:
Actually it seems to be already logged in #1598