flipperdevices/flipperzero-firmware #1554: 1-Wire DS1923

Kabouik, 2022-08-08:
### Description of the feature you're suggesting.

I was very interested to see the 1-wire contacts on the Flipperzero. At the moment, it's only meant to be used with iButtons DS1990A, but I'm wondering if it would be technically feasible to make the Flipperzero capable of interacting with their datalogger model range too (DS1923, DS1921K) to:

- Download existing data
- Possibly inject a new recording mission that would have been pre-configured in the Flipperzero storage?

We use those loggers at work and a portable and third-party way to control them and retrieve data would be nice, especially considering the numerous other features of the Flipperzero.

skotopes, 2022-09-06:
We have plans for it, but no people. I'll add it to backlog.

skotopes, 2023-03-13:
In previous release ibutton app was refactored. Basic identification is going to work for all OneWire devices.
As for data loggers it make sense to implement them in a separate external application.

closed 2023-03-13 by skotopes