flipperdevices/flipperzero-firmware #1555: Pairing code doesn't match (PS4) (BT remote plugin)

Summertime, 2022-08-08:
### Describe the bug.

PS4 says the pairing code is `000000`

Flipper says its `123456` (it changes each time obviously, point being, its not `000000`)

Pairing doesn't go through as a result.

Both devices don't allow entering/changing the pin

### Reproduction

1. Open BT Remote plugin
2. wait for device to show on PS4 devices menu
3. select the device on PS4 devices menu
4. Pairing codes appear on both devices (pairing codes don't match)
5. Press OK on the PS4
6. "The passkey might not be correct

### Target

release?

### Logs

```Text
559586 [I][BtSrv]: Bt App started
559588 [I][BtGap]: Start advertising
# Selected in devices menu of PS4
585582 [I][BtGap]: Connection parameters: Connection Interval: 9 (11 ms), Slave Latency: 0, Supervision Timeout: 500
585585 [W][BtGap]: Unsupported connection interval. Request connection parameters update
585609 [D][BtGap]: Slave security initiated
585610 [D][BtGap]: Connection parameters denied
585696 [I][BtGap]: Pass key request event. Pin: 905898
# Selected OK on PS4 with wrong pin showing
596573 [E][BtGap]: Pairing failed with status: 2. Terminating connection
596605 [I][BtGap]: Disconnect from client. Reason: 16
```

skotopes, 2022-09-06:
@gornekich can you check?

skotopes, 2023-10-30:
Looks like PS4 pairing behavior is deviating from the BLE specification.
I'll close this ticket since there is no obvious solution found and further research requires too much resources.

closed 2023-10-30 by skotopes