flipperdevices/flipperzero-firmware #1558: When running Marauder application with Wifi Dev Board, FZ reports "no databases installed"

nastea1, 2022-08-08:
### Describe the bug.

After you run the Marauder application for say 5 minutes or so, sometimes less. When you try and operate the FZ it states "No databases found on SD card.
[image]
"
Holding Left and Back for a hard reboot, resolves this every time. Not sure where to start, however it's like the database is disconnected some how. qFlipper when connected also states no databases installed when this occurs. 

### Reproduction

Turn on FZ
Run Mauarder
ScanAP
List -a
Select -a (pick one)
Run DeAuth Attack for 5 minutes or so
Exit to Main Screen
"No databases found on SD Card"

### Logs

```Text
Tell me what you want and I'll get them.
```

nastea1, 2022-08-08:
SD Card Manufacturer is SanDisk.

Astrrra, 2022-08-08:
The marauder app is not a part of our firmware. You're probably using a fork, please file the issue in their repo as we have no relation to them.

skotopes, 2022-08-08:
Do you hotplug wifi module?

3.3v ext shared with sd card, abnormal load spikes may cause DCDC to reset power, which is causing SD reset.
In general it's better to enable 5V-ext and not to hotplug wifi module.

nastea1, 2022-08-08:
> Do you hotplug wifi module?
> 
> 3.3v ext shared with sd card, abnormal load spikes may cause DCDC to reset power, which is causing SD reset. In general it's better to enable 5V-ext and not to hotplug wifi module.

Yes I do.  This was mentioned that it might be a voltage issue.  I will attempt all this, thanks for the info.

nastea1, 2022-08-08:
I can confirm that hot-plugging and not-hotplugging both work fine with 5V enabled. Thank you!

closed 2022-08-08 by nastea1

lomalkin, 2022-08-08:
Do this reproduces also in case when Flipper powered via USB?