flipperdevices/flipperzero-firmware #1561: Long-running IR signal recording/capturing

danielweidman, 2022-08-08:
### Description of the feature you're suggesting.

Right now, Flipper Zero lets you capture IR signals by entering capture mode ("Learn new remote") and then sending an IR signal. Flipper waits for a signal, then when it thinks it sees one it decides when it thinks the signal starts and ends, presumably based on timing. This can lead to some situations where long IR packets get cut off partway through (and the displayed sample count is low). It would be awesome if we could somehow set the recording going for a longer period of time where the user manually starts and stops the IR capture process, like how we can do for RAW Sub-GHz RF signal captures.

rwijbenga, 2022-08-10:
That would be a nice addition!

gsurkov, 2022-08-10:
It is an interesting idea we would like to give a go in some future. Unfortunately, this means not now.

closed 2022-08-10 by gsurkov