flipperdevices/flipperzero-firmware #1567: Boot Flipper with Pin-Locked State

Fashdey, 2022-08-09:
### Description of the feature you're suggesting.

There are currently no way to boot your Flipper in Pin-Locked State

In current Stable it is not possible to Hold the Back button until Prompted to Power Off while in Locked w/o Pin 
This only keeps screen active

What I suggest is to add the Power Off Prompt available for when Flipper is Locked with/without Pin
And when the Flipper turns back on, it should start in the state it left

E.g. for Feature Request
1. I Pin-Lock my Flipper
2. Hold Back button until Power Off Prompt
3. Powered off Flipper
4. Later I boot my Flipper
5. It is now asking for my Pin - as that is the state I left it

### Anything else?

I haven't been able to figure if this is already included in next Stable, if that is the case - close this Feature Request :)

skotopes, 2022-08-13:
There are some complications with this feature: we are unable to show shutdown window while flipper is locked. But it sounds reasonable. We'll get back to this issue after next desktop service refactoring.

Fashdey, 2022-08-17:
Alternatively it could also be accessible as an alternative shutdown, e.g. Another menu Setting or Clicking another button when having the screen - it is more the fact that I cannot power off the device if I want some kind of protection
I know it could possibly be bypassed, but there is a difference in completely open and just having a lock at least

Thanks for considering this at some point as valuable

RogueMaster, 2022-08-17:
> alternative shutdown

I see, this could be useful. Currently, if the Flipper is locked, and you reset it, it boots up locked. But you mean to lock and turn off.

RogueMaster, 2022-08-23:
@Fashdey My custom FW has this implemented now. 

@skotopes Should I PR this? 
Here is the button to [Lock W PIN + Off](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/blob/420/applications/desktop/views/desktop_view_lock_menu.c#L55) and here is [the action itself](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/blob/420/applications/desktop/scenes/desktop_scene_lock_menu.c#L71). Feel free to provide any suggested changes.

Swivelgames, 2023-05-16:
Was this also fixed by #2665?

skotopes, 2023-05-16:
@Swivelgames I think it was fixed earlier, #2665 kinda finalized locking routine and fixed bug introduced by changes in menu. But yes this issue looks to be solved as well.

skotopes, 2023-05-16:
One missing thing is power off sequence from locked state.

closed 2023-06-22 by skotopes