flipperdevices/flipperzero-firmware #1570: Flipper freezes when connected to iOS app

D4DB0DY, 2022-08-10:
[image]

[image]

D4DB0DY, 2022-08-10:
Force quitting the iOS app and rebooting the flipper makes the flipper work again. It freezes again when the app reconnects to flipper over bluetooth.

UberGuidoZ, 2022-08-10:
On connect with Bluetooth (first time), this happened instantly:

[image]

In process of debugging and gather logs.

UberGuidoZ, 2022-08-10:
[image]

UberGuidoZ, 2022-08-10:
Pay attention to time stamps to compare to log file

iOS version (iPhone 13)
[image]

Initial connecting screen
[image]

Successful connection
[image]

Crashed and disconnected
[image]

UberGuidoZ, 2022-08-10:
[Full_Crash_Log.txt](https://github.com/flipperdevices/flipperzero-firmware/files/9296521/Full_Crash_Log.txt)
[Repeating_crash_log.txt](https://github.com/flipperdevices/flipperzero-firmware/files/9296523/Repeating_crash_log.txt)
[Quick_Crash.txt](https://github.com/flipperdevices/flipperzero-firmware/files/9296524/Quick_Crash.txt)

skotopes, 2022-08-13:
Please check latest dev: there were multiple closed issues.

closed 2022-08-13 by skotopes