flipperdevices/flipperzero-firmware #1572: storage_file_seek() does not support negative offset

dummy-decoy, 2022-08-10:
### Describe the bug.

the signature for the storage_file_seek function is:
bool storage_file_seek(File* file, uint32_t offset, bool from_start);

the uint32_t type (unsigned) used for offset precludes the use of negative offset, thus we cannot go backward in the file.
also, the use of a 32 bit offset looks strange when storage_file_tell() returns a 64 bit position. 

### Reproduction

File* file = storage_file_alloc(furi_record_open(RECORD_STORAGE));
storage_file_open(file, string_get_cstr("dummy"), FSAM_WRITE, FSOM_OPEN_ALWAYS);
storage_file_write(file, "abcdefghi", 9); 
storage_file_close(file);
storage_file_open(file, string_get_cstr("dummy"), FSAM_READ, FSOM_OPEN_EXISTING);
char chr;
storage_file_read(file, &chr, 1);
assert(chr == 'a');
storage_file_read(file, &chr, 1);
assert(chr == 'b');
storage_file_read(file, &chr, 1);
assert(chr == 'c');
storage_file_seek(file, -1, false); // negative offset ?!?
storage_file_read(file, &chr, 1);
assert(chr == 'c');
storage_file_close(file);
storage_file_free(file);

skotopes, 2022-08-13:
Yep, we were having discussions on this issue. Definitely going to implement it.

DrZlo13, 2023-06-22:
Planned behavior to keep storage simple. If you need a negative offset - seek from the beginning of the file, or take furi_stream, it will do it for you.

closed 2023-06-22 by DrZlo13