flipperdevices/flipperzero-firmware #1575: Add source/input button to Infrared -> Universal Remotes -> TVs

pashdown, 2022-08-10:
### Description of the feature you're suggesting.

Having this would be extremely useful in switching a hotel TV to another HDMI port.  There are a variety of different brands in different hotels, and although the TV always has additional HDMI inputs, sometimes they give you a remote without an "input/source" button or a stream box remote with hardly anything.  Having the universal remote blast all the possible "input/source" codes would be a dream.

skotopes, 2022-08-13:
For that we need huge database of existing remotes, which we sadly don't have. We'll get back to this ticket in a couple months when we start implement new Infrared Remotes Database for mobile apps.

UberGuidoZ, 2022-09-08:
@skotopes Actually... https://github.com/UberGuidoZ/Flipper-IRDB =)

closed 2023-06-22 by skotopes