flipperdevices/flipperzero-firmware #1578: SubGHZ replay Incomplete using android app

nocomp, 2022-08-11:
### Describe the bug.

when replay a file using android app, just a small piece of the signal is broadcasted
https://ibb.co/bQP5h60
on the top original signal
bottom what is replayed

### Reproduction

turn on
launch app
finish sync to happend
choose subghz file
press send

### Target

android 12 s22 ultra

Anthony-karabian, 2022-08-11:
Just to clarify, so when you emulate the signal through the Flipper zero directly you get the full signal ? Your original signal is the signal when using the Flipper Zero directly ?

nocomp, 2022-08-12:
Yes correct

skotopes, 2022-08-13:
Are we talking about subghz raw? If yes then this problem will be handled on mobile side. We already scheduled UI/UX improvements for it.

nocomp, 2022-08-17:
hi,
just installed latest official release, and have latest android version, no changes... can t replay subghz files, just a piece of it is broadcasted even i keep fingers on send button :'(

skotopes, 2022-08-17:
Can you provide file and device info?

nocomp, 2022-08-17:
hi,
issue is with all files
here it is
https://dev.flpr.app/s#path=subghz/Store_dwn.sub&Filetype=Flipper+SubGhz+RAW+File&Version=1&Frequency=433920000&Preset=FuriHalSubGhzPresetOok650Async&Protocol=RAW&RAW_Data=1126+-130+393+-166+199+-100+201+-166+165+-166+799+-198+197+-66+329+-528+99+-132+197+-164+97+-756+229+-166+193+-562+231+-100+199+-66+99+-66+265+-198+99+-664+165+-336+165+-66+361+-132+295+-98+131+-430+133+-132+133+-198+99+-132+231+-760+859+-66+929+-134+859+-532+265+-132+99+-166+65+-430+365+-232+197+-164+197+-230+63+-132+163+-1120+131+-164+1

device info
https://media.discordapp.net/attachments/746371168449331349/1009490897936851044/Screenshot_20220817-175609_Flipper.jpg?width=279&height=597

thxx a lot!

nocomp, 2022-08-17:
as i said, if i play the file from the flipper directly everythng is fine

nocomp, 2022-08-20:
hi did you reproduced the issue?

skotopes, 2022-08-20:
Work in progress

nocomp, 2022-08-20:
thxx

DakotaMcCutch, 2022-08-25:
Same here, it goes to the flipper I can see on the flippers screen that it's playing the signal entirely for me but not actually doing anything from what I can tell. I don't have a subGHZ tool to see if it playing the signal or just looks like it

skotopes, 2022-08-26:
We'll be fixed it in next release

nocomp, 2022-08-26:
hi, thxx a lot let me know if you want me to test the pre release.
br

nocomp, 2022-08-26:
does the official rc via qflipper fix the bug?

danielemiliogarcia, 2022-08-26:
@skotopes I cant also test pre release or at specific commit

skotopes, 2022-08-31:
Sorry for delay

Long story short: signal is replayed while you press button. If you don't press long enough then signal is not going to be fully replayed. Especially when you have raw signal.

Mobile apps will improve UI/UX in next releases.

closed 2022-08-31 by skotopes

danielemiliogarcia, 2022-08-31:
@skotopes is that fixed in the latest commit? its not working for me in the latest release

skotopes, 2022-08-31:
In next release of mobile apps.

nocomp, 2022-09-01:
hi, updated mobile app + current official firmware, no changes at all :(
no sending animation on screen either :'(

skotopes, 2022-09-02:
@nocomp we have not yet released update. Also does this file works on flipper?

nocomp, 2022-09-02:
Of course!
Its been recorded from flipper, works likeba charm when sended manually