flipperdevices/flipperzero-firmware #1597: Use pre-commit for linting/formatting

miili, 2022-08-13:
### Describe the enhancement you're suggesting.

pre-commit (https://pre-commit.com/) eases and standardized the development workflow and enforces linting and formatting standards locally and in the CI. It integrates various tools:

* black
* clang-format
* codespell
* linting json/yaml/toml/md

List of other hooks https://pre-commit.com/hooks.html

Once it is set-up it can be integrated easily into the CI. Existing clang and black rules would be transferred one to one.

IMO this is a clean and reproducible solution for the maintenance. The proposed changed would replace the SCons `lint_` and `format_` commands.

atm codespell would touch 146 files, this could lead to potential conflicts.
```
 146 files changed, 327 insertions(+), 327 deletions(-)
```

skotopes, 2022-08-13:
Pre commit hooks is something that can not be enforced. Also we don't want to bloat the list of dependencies.
So to this specific framework answer is no, but we are going to improve fbt and potentially add pre-commit hooks for code format/syntax/etc check.

closed 2022-08-13 by skotopes

miili, 2022-08-13:
For future maintainability I'd strongly argue for pre-commit. It is a single dev dependency and does not bloat the codebase. This tool is what you are looking for.

@skotopes, I can work on this.

miili, 2022-08-13:
@skotopes Please reconsider.

skotopes, 2022-08-14:
Wouldn't fbt git pre-commit hook integration satisfy your needs?

miili, 2022-08-14:
Please have a look at pre-commit.com, the concept is a little different than you think. It streamlines and standardizes the codebase. I see a strong benefit for this project, since perspectively more and more ppl will get engaged.

imo extending fbt is tool abuse.

skotopes, 2022-08-14:
At this point your obsession with this tool is becoming suspicious.

I'll make one more attempt to crystalize our point of view:

- Some concepts brought in this topic we do like and will implement in future as a part of our code base
- Pre-commit hook itself is not enough
- Frameworks and tools are just ways to solve problems
- If tool is not solving our problem then we need something else
- Tight integration with build system allows us much finer control over what exactly and how we want to lint/format

Also I've took a look in the instrument itself:

- It's clearly not one dependency
- It got it's own virtual env and state
- It duplicates functions that we already have in fbt
- It also will require integration for other tools that we have 
- At this point we don't have free time on integration research and testing
- Current flow satisfies our needs

miili, 2022-08-14:
Fair enough.

Some points:
* It is one dependency which will fulfill sub dependencies in venvs. Customisation of rules is of course guaranteed!
* yes, is doublicates functionality, hence it should be removed from the entry point fbt.
* Current flow satisfy needs, but can be improved. Therefire this proposal.

Thanks for all your work! Cannot wait for my device. :)