flipperdevices/flipperzero-firmware #1598: Support NFC A 8-bytes UID

red-avtovo, 2022-08-13:
### Describe the enhancement you're suggesting.

I just scanned a card and I noticed that it is not 7, but 8 bytes uid instead. 
I can save it, but I can't emulate it
[image]

miili, 2022-08-13:
I count 8 bytes in the screenshot.

red-avtovo, 2022-08-13:
Correct. As I said earlier, I can read it and save, but can't emulate it. 
Once I open the saved card it says that the data is corrupted

skotopes, 2022-08-13:
Hi, there are some hardware limitations that we have on NFC chip side. We are currently researching what we can do, but most likely 8 byte UIDs are not coming any time soon.

pllopis, 2022-08-13:
> Hi, there are some hardware limitations that we have on NFC chip side. We are currently researching what we can do, but most likely 8 byte UIDs are not coming any time soon.

Hello, can you elaborate further on these limitations? I was just reading through the NFC layer to see how to add experimental NFC-V cloning/emulation support.

skotopes, 2022-08-13:
@pllopis there are some operations like anti-collision that are off-loaded to ST25R3916.
It can only handle 4 or 7 byte UIDs for NFC-A. In theory it can be done in software, but that where things become very complex, especially with timings.

Hardware emulation support for NFC-V is not in this chip. So you'll have to do it fully in software. May be a challenge.
Also please check ST25R3916 datasheet and other projects, we haven't been investing much time into V. Could be that someone solved this problem.

pllopis, 2022-08-14:
Thanks for the answer @skotopes! Indeed I've seen in the data sheet that NFC-V emulation is not supported by the hardware. Anti collision might indeed get tricky to handle in software, and that would need to be a low level implementation close to the data link layer, I guess.

Anti collision/inventory/discovery aside, does the currently FURI implementation provide enough abstractions that it would be a matter of working at the APDU level for emulating NFC-V, or would there be more missing bits at lower levels such as transport or data link layers? I'm just starting to look at the source, and I wonder if there's further documentation about this (several GH issues reference a FURI wiki, which seems to no longer exist).

gornekich, 2022-09-07:
Hello @red-avtovo !
I think that this 8-bytes UID tags are actually NFC-B and Flipper incorrectly recognized them as NFC-A. Could you please try to read this card with latest release (0.66.1 at the time of writing) and write here the result?

red-avtovo, 2022-09-07:
Sadly I've checked out from that Hotel, but I will try soon with another one

closed 2022-12-01 by skotopes

alvarotorijano, 2023-01-07:
i've tried with an ISO15693 recognized as NFC-V and shows the 8-byte UID. Able to read the UID but not the info and cant emulate. I've been able to fully read the tag with the proxmark3. Do anybody knows how is this going on the firmware roadmap?

KondratokGK, 2023-07-09:
I have some iso15693 from ti. It`s for "factorial" intercom. I can read this with nfc tools and custom commands.

yash95bh, 2023-09-30:
When I try to read a card, it only reads the UID and not the info/data of the card. Hence when I emulate, it only shows the option to emulate UID and not the data inside. How do I fix this?

How do i fix the cannot parse file issue as well?

skotopes, 2023-10-12:
@yash95bh can you provide more details? Card type, read logs? Also new NFC stack is coming, please try it too https://github.com/flipperdevices/flipperzero-firmware/pull/3050

yash95bh, 2023-10-13:
When I first got the flipper, I read a hotel card where I work and it read as follows:
- Mifare Classic 1K - NFC A with UID number
- It stored the data and I emulated it, it successfully opened the doors

I updated the saved card from the flipper and this is how it reads now:
UNKNOWN ISO TAG
ISO 1443-3 (NFC-A)
UID: 00 00 00 00
ATQA: 00 00 SAK: 08

Note: The UID has changed to 00  now
When I read the same card under magic card reader, it successfully detected it. Somehow the UID changed . I have attached a screenshot of it

[image]

Later, I made another master key (we use Onity card maker), tried to save it > It saved and was emulating as well but it didn't open any locks.

-----------------------
Another issue with NFC

I download HEX editor and changed the codes. I copied the file from the desktop to the SD card and it shows an error saying CANNOT PARSE THE FILE

[image]

Am I pasting the file wrong? Do I need to buy another device such as NFC cloner / Duplicator? 

ALL I WANT IS TO READ A CARD, CHANGE DATA IF REQUIRED , WRITE IT ON DIFFERENT CARDS (MAGIC CARDS) AND SUCCESSFULLY EMULATE IT.

Your help is much appreciated.