flipperdevices/flipperzero-firmware #1600: iclass support

d0minikkkk, 2022-08-14:
### Description of the feature you're suggesting.

Currently, the firmware does not seem to provide iclass support. I found an iclass plugin under an alternative firmware for flipper on [another repository](https://github.com/Eng1n33r/flipperzero-firmware). Is it possible to merge the iclass/picoclass plugin into the official firmware? 
Code for the plugin can be found [here](https://github.com/Eng1n33r/flipperzero-firmware/tree/dev/applications/picopass).

skotopes, 2022-08-14:
Technically it's already merged, just not built by default. It will be included as external application as soon as we release apps SDK.

closed 2022-08-14 by skotopes

d0minikkkk, 2022-08-14:
is there instructions to build for the feature? I can't find anything regarding this under build instructions.

skotopes, 2022-08-14:
`./fbt flash --extra-int-apps=picopass`