flipperdevices/flipperzero-firmware #1603: USB key storage support

Altair31, 2022-08-15:
### Description of the feature you're suggesting.

Nowadays, we can only send files through USB by using the qFlipper program.
It would be great if the Flipper by default is recognized as a regular USB storage so we can use it on any computer or car embedded audio system to read/write files on SD card.

skotopes, 2022-08-15:
Hi. It's going to be painfully slow because we use SD card in SPI mode.
There were multiple discussions and couple prototypes made, no good.

non-bin, 2022-08-16:
I know it's slow, but at the moment it's inconvenient too if you don't have a card reader

skotopes, 2022-08-16:
It’s going to be unconvintly

closed 2022-08-16 by skotopes

non-bin, 2022-08-16:
Right now you have to go through qFlipper making it even worse, could you point me to the commits where the test versions are?

skotopes, 2022-08-16:
It's actually faster in qflipper, because it uses high level protocol with minimum overhead.
Exposing block level access to the system brings a lot os small transactions which makes it slow. Plus modern OS tend to do a lot of background work like indexing and etc which makes it even worser.

skotopes, 2022-08-16:
We really did some test and results are not satisfying.

non-bin, 2022-08-16:
Ok, thanks for the detail :)