flipperdevices/flipperzero-firmware #1607: Profiling in SubGHz section

albertoglezc, 2022-08-15:
### Description of the feature you're suggesting.

It would be nice to have profiles in SubGHz section with a definición file like

Name:frequency,modulation,<raw/noraw>

If some remote controls or systems uses a known frequency it would be mire easy (instead of using first Frequency Analizer) and after that try to use modulation to capture data or try raw..

skotopes, 2022-09-06:
Partially we already have it. We'll revise system in future updates

skotopes, 2022-10-16:
We've made some research and tests we'll stick to currently implemented chip-specific profile system.
Other options requires complex abstractions and chips that got fully open documentation which one of our vendor doesn't provide.

closed 2022-10-16 by skotopes