flipperdevices/flipperzero-firmware #1611: Filed to emulate NFC

gushmazuko, 2022-08-16:
### Describe the bug.

NFC emulation has stopped working since the [0.63.3](https://github.com/flipperdevices/flipperzero-firmware/releases/tag/0.63.3) update. In the release notes you can see `NFC: completely redesigned application, improvement ...`, I guess the problem is here.
I rolled back to the previous firmware [0.62.1](https://github.com/flipperdevices/flipperzero-firmware/releases/tag/0.62.1) and the emulation works fine. Then I updated to [0.63.3](https://github.com/flipperdevices/flipperzero-firmware/releases/tag/0.63.3)  again, the emulation doesn't work again. Same with [0.64.3](https://github.com/flipperdevices/flipperzero-firmware/releases/tag/0.64.3) & `dev b3d7583f`.

After each update I read the tag again.

### Reproduction

1. Read Mifare DESFire tag
2. Save
3. Emulate

gornekich, 2022-08-18:
Hello @gushmazuko , thanks for report, working on it.
Do you have the problem with emulating DESFire tag UID?

gushmazuko, 2022-08-18:
> Hello @gushmazuko , thanks for report, working on it. You have the problem with emulating DESFire tag UID, am I right?

Yes exactly.

elgumso, 2022-08-18:
I am actually having the same issue but with saving/emulating a Mifare Classic UID. I can emulate the UID from my capture file I took whilst on a previous firmware version, but a new read/save of the same card on the current firmware results in no "Emulate UID" option, just "Emulate" which doesn't work.

incyi, 2022-08-21:
I have the same problem. Trying now to scan the emulated nfc from the flipper zero with my android phone.

mgrybyk, 2022-08-25:
Same issue with every Miflare Classic 1K that used to work before, example
```
ISO 14443-3 (NFC-A)
UID: 04 30 66 62 5B 55 80
ATQA: 00 44  SAK: 08
Keys Found 32/32
Sectors Read 16/16
```

gornekich, 2022-08-29:
Hello @gushmazuko . Fix in this #1643 PR, soon will be in dev

theblackhole, 2022-08-29:
> Hello @gushmazuko . Fix in this #1643 PR, soon will be in dev

Hello @gornekich, do you think this will also fix  #1468 (Mifare Classic 1K) or is it another issue than this DESFire one ?

gushmazuko, 2022-09-01:
> Hello @gushmazuko . Fix in this #1643 PR, soon will be in dev

It works fine, thank you!

closed 2022-09-06 by skotopes

netdoggy, 2022-10-09:
@gornekich
Please tell me, should Miflare Classic 1K be emulated with a flipper completely or only UID?

If I take the saved Miflare Classic 1K file (from the intercom), together with the keys in the sectors, and click emulate, my android does not react to it in any way. At the same time, it perfectly reads the original chip.