flipperdevices/flipperzero-firmware #1614: Choosing specific nfc type to read

paulpaul115, 2022-08-17:
### Description of the feature you're suggesting.

Since phones might generate more than one type nfc signal(e.q, Japan's phones generally have NFC-B and NFC-F active at same time), It is very hard to capture the specific one from Flipper Zero

GMMan, 2022-08-20:
Are you talking about capturing from phones specifically, or a card? You cannot capture signals from phones aside from emulating a particular card to the phone.

paulpaul115, 2022-08-21:
to be specific, phones give out NFC-B signal for Visa paywave and google pay, etc., and NFC-F(Felica) for subway pass(Suica and others prepaid card issued all around Japan). Phones have fixed NFC-F IDm(a dedicated chip in phone, and do nothing with NFC-A/B). I also tried emulating NFC-A card by phone, and seldom success because of phone spamming NFC-B signal at same time

skotopes, 2022-09-06:
Yep, we do plan to add more manual actions. F/B is one of them

dogtopus, 2023-04-22:
Anyone working on this currently?

This is required to get `Read Specific Card Type` to work correctly once #2093 is done.

skotopes, 2023-06-22:
Yep, arrived some time ago: NFC -> Extra Actions -> Read Specific Card Type

closed 2023-06-22 by skotopes