flipperdevices/flipperzero-firmware #1617: Issues w/ `includePath` on mac

ppseprus, 2022-08-17:
### Describe the bug.

Yesterday, I was still able to experiment with an application Im working on. Writing code, flashing it onto my Flipper...
As of today, I've been having all kinds of issues w/ `#include`, but only on the new folder I have created for my application.

I have issues w/ the following lines
* `#pragma once`
* `#include <furi.h>`
* `#include <furi_hal.h>`
* `#include <furi_hal_resources.h>`
* etc.

Have I missed something? What am I missing?

### Reproduction

Not sure what happened.
I rolled back to `0.64.2`, thinking maybe its the latest `dev` that has issues — nothing.
My `.vscode` folder is setup with the `./fbt vscode_dist` command
I stashed all my changes, navigate to `0.64.2` (`7a672038`) and simply added the [Hello World plugin](https://github.com/mfulz/Flipper-Plugin-Tutorial) and I got the same results.

[image]

### Logs

```Text
[8/17/2022, 11:17:00 PM] "hello_world.c" not found in "${workspaceFolder}/build/latest/compile_commands.json". 'includePath' from c_cpp_properties.json in folder 'flipperzero-firmware' will be used for this file instead.
[8/17/2022, 11:17:50 PM] "hello_world.c" not found in "${workspaceFolder}/build/latest/compile_commands.json". 'includePath' from c_cpp_properties.json in folder 'flipperzero-firmware' will be used for this file instead.
```
```

DrZlo13, 2022-08-23:
Have you compiled this code at least once?
compile_commands.json is generated on first build, so you need to build the code before processing the file with vscode.

skotopes, 2022-09-06:
Please reopen issue if problem persist.

closed 2022-09-06 by skotopes