flipperdevices/flipperzero-firmware #1630: Flipper Auto Shutdown after inactivity

ase1590, 2022-08-20:
### Description of the feature you're suggesting.

An inactivity auto-shutdown feature would be nice to have to prevent the battery from running down after being left on for days. 

Having a customizable auto-shutdown inactivity timer available in the Flipper settings menu would be ideal in order to help prevent the battery from totally discharging while flipper is stored out of sight in a drawer :)

grievertime, 2022-08-20:
+1, would be a nice feature to have

RogueMaster, 2022-08-23:
How is this different than the auto-lock feature?

grievertime, 2022-08-24:
> How is this different than the auto-lock feature?

power consumtion, shutting down flipper allows it to last months without charging.

azaitsev, 2022-08-28:
https://github.com/flipperdevices/flipperzero-firmware/pull/1647

Ethanol0001, 2023-03-03:
Having this feature as an option would be nice. However, the flipper's battery consumption is low enough when not being used that it can last weeks on a single charge and with normal use, around a week. To add onto that, I have found issues with the xp clock not progressing when in transit mode(fully powered down). In general it would probibly be best to just leave it to have auto-lock which is an option in the settings alredy.