flipperdevices/flipperzero-firmware #1632: Different SAK after NFC Dump ?

Foul, 2022-08-21:
### Describe the bug.

i dumped my door bagde with NFC Tools on my phone :
[Door.dmp.txt](https://github.com/flipperdevices/flipperzero-firmware/files/9389697/Door.dmp.txt)

and if i dump the same badge with flipper :
[Door.nfc.txt](https://github.com/flipperdevices/flipperzero-firmware/files/9389698/Door.nfc.txt)

SAK is not the same !?! 88 on Original Dump, 08 with Flipper

### Reproduction

make a dump ith proxmark, NFC tool or any NFC dump tool..
Check the original SAK

Dump the same thing with Flipper
Check the SAK.

dorianboulch, 2022-09-20:
Same thing happen to me

AkechiShiro, 2022-10-30:
Could you provide the tag's brand ?? This doesn't happen with any tags I've had on my hands so far.

dorianboulch, 2022-10-30:
I think this is related to #1468

Astrrra, 2022-12-08:
Can you still reproduce this issue?

Astrrra, 2023-03-07:
Closing this as inactive, please re-open if the problem persists.

closed 2023-03-07 by Astrrra