flipperdevices/flipperzero-firmware #1638: Option to Detect Driver/Protocol on Saved files

ppseprus, 2022-08-22:
### Description of the feature you're suggesting.

Save [this](https://github.com/flipperdevices/flipperzero-firmware/pull/1625#issuecomment-1221525241) comment yesterday, by @tkerby and I wanted to follow up on it. I did not see any Feature Request like that, so I decided to open it.

The list of known drivers and protocols change by each firmware version, and it is also subject to customisation.

An option to "Detect Driver/Protocol" for raw files would be really practical.

tkerby, 2022-08-22:
Interestingly you can process a raw file using the CLI but not the UI. Would be great to integrate post processing in the UI

ppseprus, 2022-08-22:
Wow! I did not realise that! Do you have a docs on that?

qistoph, 2022-08-23:
@ppseprus, There are brief docs in the `subghz` command on the command line.

To decode a raw recording use `subghz decode_raw <FILE>`

ppseprus, 2022-08-23:
Awesome! Thanks!

Still, I think a UI feature would be practical.

Skorpionm, 2022-08-27:
https://github.com/flipperdevices/flipperzero-firmware/pull/1667

closed 2023-03-13 by skotopes