flipperdevices/flipperzero-firmware #1641: Chamberlain 9-bit capture not replayable

ESurge, 2022-08-23:
### Describe the bug.

Capturing a Chamberlain 9-bit signal allows me to save to file, but upon replaying the signal, I receive "Transmission is blocked" message.

I can record and replay the signal with Read RAW.

Also, if I try to open the saved file, I receive "Cannot Parse File" message.

Opening `log` in CLI when attempting to open the file or replay the captured signal, I receive "Wrong number of bits in key".

### Reproduction

1. Start Sub-GHz > Read with proper settings on FlipperZero
2. Send & Capture Chamberlain 9-bit signal
3. Attempt to resend captured code

### Logs

```Text
[E][SubGhzProtocolChamb_Code] Wrong number of bits in key

File contents:
Filetype: Flipper SubGhz Key File
Version: 1
Frequency: 433920000
Preset: FuriHalSubGhzPresetOok650Async
Protocol: Cham_Code
Bit: 9
Key: 00 00 00 00 00 00 01 55
```

### Anything else?

Tested on Release 0.64.3, RC-1 0.64.3, and Dev 57328761

Region: Canada

Skorpionm, 2022-08-24:
i will check it, it should work

Skorpionm, 2022-08-24:
https://github.com/flipperdevices/flipperzero-firmware/pull/1650
fix

ESurge, 2022-08-24:
Tested on latest clone of dev branch (ce7b9437) and this works now.

Thanks.

closed 2022-08-24 by ESurge