flipperdevices/flipperzero-firmware #1646: MF Classic - switching dictionary priorities

zviratko, 2022-08-23:
### Describe the enhancement you're suggesting.

Hi,
I think NFC/Read shuld first use the default dictionary and only then switch to the (larger) user dictionary. It takes about 2 hours to read a MF Classic card "by default" without skipping the user dictionary attack, few seconds with the default dictionary. (I also feel there has been regression in there, it used to be much faster, but it probably depends on what "default" keys the card uses).

Can the order of the dictionaries be switched? I think it makes sense to start with the smaller but much more probable library of default keys and only then go to the much larger user library.

Thanks

Foul, 2022-08-25:
When i'm looking for keys, i'm first using **known** of them in my own user dic.. then search for **unknown** in the larger default dic...
When all keys are known, just move them from user to default dic...
i think it's faster using already known keys with user dic, and then, other dictionary.
Just my own opinion.

zviratko, 2022-08-25:
To be honest I'm not sure how this works in the OG firmware as I'm using the RogueMaster one. In there (and I think also in Unleashed) there are more keys added to the "user" dictionary.
In practice, most MF cards I see are using the default or well-known keys, so using the default dictionary works instantly, but using the user one chugs for hours and I have to skip the user dictionary everytime.
I understand your point, though. Maybe what we need is a "default" dictionary with few keys that work on 98% of cards, a "known-keys" dictionary that would take a few hours and a "user" dictionary for non public but user owned keys. Preferably with configurable order. Would that make sense?

Foul, 2022-08-25:
Most common Keys are 00000000 and FFFFFFFF and can be put in both of dic
After 1 dic with Full keys.. and user dic with the most used by user... seems legit ;)

gornekich, 2022-09-07:
Hello @zviratko 
As @Foul mentioned, the correct way to read MFC tags is to first try User dictionary, which is fully controlled by user and should  be "fast" operation. It's up to the user what keys he/she wants to add to user dictionary and official firmware doesn't add any keys here. It is assumed that user will add keys in User dict after running attacks, such as reader attack, that is partly supported right now. The order of dictionary will not be changed.

closed 2023-03-13 by skotopes