flipperdevices/flipperzero-firmware #1659: Subghz came bits error

rmukeng, 2022-08-25:
### Describe the bug.

I have a file with 13 bits when I try to run the file, it shows parse error this has only recently started happening 

### Reproduction

Run subghz sub file that points uses came that contains 13bits

Parse error is then shown

ESurge, 2022-08-25:
This looks related to my issue #1641, and it was fixed in #1650 

You might need to wait for the change to be put into the next Release or Release Candidate. Alternatively, use the latest Dev build [Aug 24 - ce7b9437](https://update.flipperzero.one/builds/firmware/dev/flipper-z-f7-full-dev-24082022-ce7b9437.dfu)

rmukeng, 2022-08-25:
I have tried what's mentioned on #1650, unfortunately this does not resolve the issue with my bits being 13

Skorpionm, 2022-08-25:
and will not decide on the protocol for CAME or 12 or 24 bits

rmukeng, 2022-08-25:
I am trying a sub file which shows 13bits but it just shows an unable to parse error message

Skorpionm, 2022-08-25:
where did you get it? turning to the one who created this file

rmukeng, 2022-08-25:
They are from

https://github.com/UberGuidoZ/Flipper/tree/main/Sub-GHz/Gates/Road_Arm_(UK)

They previously worked before the came change

Skorpionm, 2022-08-27:
so you need to ask the question of whoever did them wrong

closed 2022-08-27 by Skorpionm

rmukeng, 2022-08-27:
Thanks looks like the came protocol only supports 12bits no more no less change it to 12 and it parses fine