flipperdevices/flipperzero-firmware #1688: Legic Prime support

Builderhummel, 2022-08-31:
### Description of the feature you're suggesting.

Legic Prime is a similar standard like Mifare and it is common in europe.

It would be helpful to be able to read and emulate the legic prime cards just like mifare.

(The algorithm is also available in the Proxmark3 source code, like the Mifare one)

### Anything else?

The flipper project is really interesting. Legic Prime support would make it a lot more practical.

david-bla, 2022-09-01:
Legic Prime is really common here in europe.
There is a big market for cheap access systems and the fact that it is totally reversed (broken) does not mean security consultants don't recommend it anymore.
This would be a great addition for the flipper :+1: .

I am not sure but seems like the hardware should be capable of it.
For reference see the implementation of proxmark3 like @Builderhummel mentioned:

https://github.com/Proxmark/proxmark3/blob/master/client/cmdhflegic.c

and its implementations:

https://github.com/Proxmark/proxmark3/blob/master/armsrc/legicrf.c
https://github.com/Proxmark/proxmark3/blob/master/common/legic_prng.c

BuckarooBanzay, 2022-09-01:
can't really help on the implementing front here but i have the hardware to test it, just ping me up

skotopes, 2022-09-06:
We'll come back to this after RFID rewriting in C

Builderhummel, 2022-09-12:
@skotopes Thanks. The Frequency tag is wrong tho, since Legic Prime uses 13,56 MHz.

panki27, 2022-09-30:
Worth noting that the Proxmark sources use its built-in FPGA, so adapting the code is going to be challenging.

Here is a talk about how the Legic protocol was reversed: https://fahrplan.events.ccc.de/congress/2009/Fahrplan/events/3709.en.html
The linked slides have a lot of detail.

Tetris81, 2023-02-05:
is there any progress?

Tetris81, 2023-02-05:
@skotopes is there any progress?

skotopes, 2023-02-05:
@DrZlo13 wanna take a look?

skotopes, 2023-02-05:
> @skotopes is there any progress?

Can you attach RFID RAW records here?

Tetris81, 2023-02-05:
@skotopes i have one but is my work card
But I found a patch for LegicPrime on the site, maybe that will help:

https://itooktheredpill.irgendwo.org/2010/mensakarte-entschluesselt/
https://itooktheredpill.irgendwo.org/2010/legic-karten-beschreiben/  "Den eigentlichen Patch gibt es hier: legicfastreader.diff"
>> legicfastreader.diff 
[legicfastreader.diff.txt](https://github.com/flipperdevices/flipperzero-firmware/files/10610861/legicfastreader.diff.txt)

I don't know anything about programming but it looks like c+

skotopes, 2023-02-05:
We still need some keys or records. Is there any place where we can order it online?

Philipp2683, 2023-02-05:
@skotopes Do you have discord?

panki27, 2023-02-06:
One of their obscurity strategies is afaik to (almost) only sell B2B. But after some searching, I found this shop in the UK that doesn't require you to request a quote:
https://www.windmill-computing.com/products/view-one/pn/BDG-ISO-LEGIC-P1024/page/smart-cards

skotopes, 2023-02-06:
@panki27 @Astrrra will research this protocol when she get time

Astrrra, 2023-03-07:
HI! Support for Legic Prime is technically possible for us, we will implement it in the future.

achtnullzwei, 2023-04-03:
@Astrrra I suspect I have some legic cards and I did a raw read from rfid menu. Lmk if the files can be of help. Ability to read/emulate those would be awesome.

parsley, 2023-06-19:
Hi!
It looks like this FR belongs to this forum post:
https://forum.flipperzero.one/t/legic-prime/14159
Right?

devBioS, 2023-11-05:
Legic Prime is technically all about timing, there are good resources about problems with timing and their solutions while emulating tags on the proxmark forum.
The tag raw values are "encrypted" with a XOR value also present on each card.
Most importantly they use custom non-standard crc8 and crc16 generators to validate values present on the card which need to be calculated on the fly if emulating cards.

Also here the proxmark forum is a very good source of knowledge for prime tags..
Raw support for Legic Prime Tags would be really awesome, where one could build an app upon :)

Boringsapiens, 2024-04-15:
Hey @gornekich @skotopes any updates?

skotopes, 2024-04-26:
Not really

skotopes, 2024-12-02:
Nope, it is in the queue, but with low priority. There is no specific date that I can promise to you guys. Really sorry.

Nomcon, 2024-12-27:
In case the RFID raw values are still relevant.
I just captured the ones of my Legic chip.
I had to rename them to tip them so in order to upload, do with that what you will.
[RfidRecord.zip](https://github.com/user-attachments/files/18264028/RfidRecord.zip)

TudbuT, 2026-10-13:
What does low priority even mean in this case?

Msprg, 2026-10-13:
# Update: working LEGIC Prime prototype (read + emulate)

I have LEGIC Prime reading and emulation working on a stock Flipper Zero, with no extra hardware. See: https://github.com/Msprg/flipperzero-firmware/tree/legic-dev

What's been tested so far:
- Reading a real card: a MIM256 card dumped 5 out of 5 times, byte-for-byte identical to Proxmark3 hf legic dump.
- Reading simulated cards: MIM22, MIM256 and MIM1024 from Proxmark3 hf legic sim.
- Emulation: the Flipper emulating a MIM256, read back byte-exact by Proxmark3 (hf legic info / hf legic dump). Not yet tried against a real LEGIC reader.

To try it (CLI only for now, no GUI): build and flash the branch (./fbt flash_usb_full), then in the Flipper CLI:
```
nfc
dump -p legic -f /ext/nfc/legic.nfc -t 15000
emulate -f /ext/nfc/legic.nfc
```
Technical notes:
- The ST25R3916 has no LEGIC mode (because of course it doesn't). Therefore RX uses its 212 kHz subcarrier-stream receiver, re-armed after every run of 1-bits.
- Real cards drop the session if the reader's next frame doesn't follow their reply within about 220 µs, which is less time than it takes to decode the reply. So the next frame is prepared in advance and sent straight from the receive path.
- Emulation runs the chip in transparent mode and generates the reply subcarrier on the MCU.

However:
- Only one real card has been tested.
- Debug logs are written to /ext/legic_*_dbg.log. Please attach them if something fails.

---
> [!TIP]
> This is currently just a prototype, still very much in development!

> [!TIP]
> Reports from people with other cards (especially MIM22 or MIM1024) or with real LEGIC readers would help a lot. Please open a dedicated issue on my fork for these purposes. Thank you~!

> [!NOTE]
> AI-assisted development

Msprg, 2026-10-16:
> [!TIP]
> I've opened #4468 with LEGIC Prime support (MIM22, MIM256 and MIM1024):
---
- **Read** in the NFC app (card info, and the decoded segments under More) or with `nfc dump -p legic`
- **Save** to `.nfc` and **emulate** a saved card (NFC app or `nfc emulate`)
- **Write** from the CLI: `nfc legic wrbl -o <offset> -d <hex>`, and read back with `nfc legic rdbl -o <offset> -l <length>`. Bytes 0-4 can't be written. The DCF (bytes 5-6) needs `--danger`, because changing it can't be undone on a real card. Writing from the NFC app isn't offered.

Reading and writing were tested on a real MIM256 card, with dumps byte-exact against Proxmark3 `hf legic dump`. They were also tested against Proxmark3's `hf legic sim` for all three card sizes. Emulation was tested with a Proxmark3 as the reader, **but not yet with a real LEGIC reader**. If you have one (a door reader, a canteen or payment terminal, etc.), it would be very helpful to know whether it accepts an emulated card. Reports from other LEGIC Prime cards, especially MIM1024 and MIM22, are welcome too. If you try writing, please only do it on cards you own and can afford to change.

You can try it by building the PR branch (`Msprg:legic-prime-pr`) and flashing it, e.g. `./fbt flash_usb_full`.

broglep, 2026-10-16:
Successfully tested with two different LEGIC reader part of a KABA legic system (based on MIM1024). Could read the key and then emulate to unlock doors

Msprg, 2026-10-16:
> 26\. Successfully tested with two different LEGIC reader part of a KABA legic system (based on MIM1024). Could read the key and then emulate to unlock doors

That's amazing! Glad to hear the emulation's actually working properly!

Msprg, 2026-10-17:
@broglep I have pushed some updates that were requested by the maintainers in my PR. Feel free to test the updated code to make sure the readers still read from flipper correctly even on the latest commit since the emulator code was changed

Thank you and hope the pr gets merged soon 🙂

broglep, 2026-10-20:
I have retested emulation and it is still working with the readers

stevenrezzonico, 2026-11-12:
I confirm it works fine on a MIM1024 card