flipperdevices/flipperzero-firmware #1691: Add UI option to clear NFC cache

ThePowerOfDreams, 2022-09-01:
As a feature request, please consider adding a UI option to clear the NFC cache so that expanded dictionaries can be tried against cards that were previously read.

gornekich, 2022-09-07:
Hello @ThePowerOfDreams 
Why do you need to delete nfc cache for that? Saved keys caches allows to read already known sectors, and if card is still not fully read, dictionary attack starts after that

ThePowerOfDreams, 2022-09-07:
>Saved keys caches allows to read already known sectors, and if card is still not fully read, dictionary attack starts after that

That's not the behaviour I'm seeing with a Mifare card with only some keys found; after one second or so, Flipper shows me 26/32 keys and 16/16 sectors, but will not actually retry it even when I tap left for Retry.

gornekich, 2022-09-07:
Got it, it's a bug. I don't see why we need to delete key cache, but we definitely have to start dictionary attack unless not all keys found. Will be fixes soon

Astrrra, 2022-12-08:
The core issue was fixed in #1868. Please re-open the issue if that was not enough.

closed 2022-12-08 by Astrrra