flipperdevices/flipperzero-firmware #1700: Recover >8MB of ROM

csBlueChip, 2022-09-04:
### Describe the enhancement you're suggesting.

If you run `grep  '\[[0-9]'  lib/u8g2/u8g2_fonts.c  |  sed 's/.*\[\([0-9]*\)\].*/\1/'  |  wc -l` you will see there are currently 1689 fonts installed.  
This costs `grep '\[[0-9]'  lib/u8g2/u8g2_fonts.c  |  sed 's/.*\[\([0-9]*\)\].*/\1/'  |  paste -sd+ -  |  bc` == 10,586,778 bytes  ~= 10.1 MB of ROM space

Given the API only actually gives you access to FOUR of these fonts, would it be worth curating this list to recover a significant amount of ROM ...Even from a wear-levelling perspective it makes sense to avoid excessive/needless ROM writes.

EG. Just getting rid of these fonts would recover 3.3MB of ROM
```c
u8g2_font_unifont_t_japanese3[161658]
u8g2_font_wqy12_t_gb2312[208526]   
u8g2_font_wqy12_t_gb2312a[111517]  
u8g2_font_wqy12_t_gb2312b[120553]  
u8g2_font_wqy13_t_gb2312[227679]   
u8g2_font_wqy13_t_gb2312a[121482]  
u8g2_font_wqy13_t_gb2312b[131121]  
u8g2_font_wqy14_t_gb2312[251811]   
u8g2_font_wqy14_t_gb2312a[134054]  
u8g2_font_wqy14_t_gb2312b[143653]  
u8g2_font_wqy15_t_gb2312[277226]   
u8g2_font_wqy15_t_gb2312a[147711]  
u8g2_font_wqy15_t_gb2312b[158881]  
u8g2_font_wqy16_t_gb2312[318389]   
u8g2_font_wqy16_t_gb2312a[169444]  
u8g2_font_wqy16_t_gb2312b[182450]  
u8g2_font_b12_t_japanese3[111424]  
u8g2_font_b12_b_t_japanese3[110441]
u8g2_font_b16_t_japanese3[161307]  
u8g2_font_b16_b_t_japanese3[165572]
```

And, although there is (ironically) very little to be gained in terms of ROM, the 78 and 92 pixel high fonts are too big to be displayed!

### Anything else?

In the unlikely even that a developer wants access to a specific font rejected by the curator, they could easily cut and paste it from `u8g2_fonts.c` in to their code locally.

Furthermore, to facilitate using additional fonts, I propose the addition of :
```c
void  canvas_set_font_u8g2 (Canvas* const canvas,  const uint8_t* u8g2Font)
{
    u8g2_SetFontMode(&canvas->fb, 1);  // no idea - but canvas.c does it <shrug>
    u8g2_SetFont(&canvas->fb, u8g2Font);
}

```

skotopes, 2022-09-04:
These fonts are already automatically stripped by linker.

closed 2022-09-04 by skotopes

GMMan, 2022-10-11:
As a side note, there's only 1MB of flash on the MCU total anyway, and about a quarter of that's reserved for the wireless stack. It's not logical to have working firmware if there's more space taken up by fonts than there is space in flash.