flipperdevices/flipperzero-firmware #1711: Wrong pin sd format

FalsePhilosopher, 2022-09-06:
### Description of the feature you're suggesting.

Make an option in the settings menu to enable 3 or 5 wrong pin attempts and after the failed 3 or 5 attempts it will enter a mode where it doesn't accept button inputs, formats the sd card, fills it with 1's and 0's, then disables the start up function and turns off. If you wanted to turn it back on you would have to enter dfu mode with the button combo and restore the FW.

skotopes, 2022-09-06:
Hi.

Doesn't make much sense. What will stop attacker from copying sd-card first? Also our sdcard exchange speed is only 500kbps, secure format will take eternity. Internal storage self erase is also kinda possible but too much hassle.

closed 2022-09-06 by skotopes

FalsePhilosopher, 2022-09-06:
> Hi.
> 
> Doesn't make much sense. What will stop attacker from copying sd-card first? Also our sdcard exchange speed is only 500kbps, secure format will take eternity. Internal storage self erase is also kinda possible but too much hassle.

It's not made for an attacker, it would be targeted at if some normie stole your FZ. An attacker would know to just pop out the SD card, but not some normie trying to play with your stolen flipper. It's a better idea to wipe enough of the internal storage to quasi soft brick the device. An just a basic format on the sd card then, it wouldn't stop somebody that knows data forensics. But it would stop any normie.

skotopes, 2022-09-06:
Very niche thing that will be half working. Answer is still no.