flipperdevices/flipperzero-firmware #1713: SubGHz regdb support for non-US/EU/JP regions

perillamint, 2022-09-07:
### Describe the enhancement you're suggesting.

Currently, Flipper Zero firmware does not support individual regulatory rules per country, but in four categories. For countries in those three categories, it is possible to transmit on whole frequency range legally permitted on country.

However, in Rest of World categories, it is quite troublesome to transmit on some band even if it is legally permitted to transmit, and it forces users in RoW to use "unlocked" firmware to fully utilize its ability legally.

To address this problem, I think there should be more detailed database and regulatory regions provided to users.

I have some ideas to implement this in the flipper firmware:

a) put everything in struct as currently flipper zero does.
b) detach regdb file to sd card and validate it using digital signature like [Linux wireless-regdb](https://wireless.wiki.kernel.org/en/developers/regulatory/wireless-regdb#the_regulatory_database) does

mattimac, 2022-09-07:
We should provide 3 versions for ITU regions https://en.wikipedia.org/wiki/ITU_Region cause frequencies differ according to those. for example 915MHz is ITU region 2 only. To make it easy there should be easy ITU region selection.

Ads-Decadence, 2022-09-20:
Good day peeps,   I have an issue I live in Jersey Channel Islands United Kingdom, flipper hardware provisioned marks me as JE (this is correct) but only gives me radio frequencies available to R03, not the UK. 
I belive any users in Gurnsey (or the other Channel Islands) will have the same issue

skotopes, 2022-10-16:
In many countries local regulations override ITU regulations. Currently we open bands based on official bands documentation by appropriate country authority.

One more limitation is that we are opening frequencies that are covered by our current certificates.

I'm planning to open bands database soon and we are going to change to something similar to wireless-regdb in future products.

closed 2022-10-16 by skotopes