flipperdevices/flipperzero-firmware #1714: Japan Regional lock (Japan's Sub-GHz regulation)

paulpaul115, 2022-09-07:
### Description of the feature you're suggesting.

According to _**Telecommunications Bureau of the Ministry of Internal Affairs and Communications**_ in Japan, there are two categories of Radio Station called [Extremely Low Power Radio Station] and [Specified Low-Power Radio Station] user did not require license to operate, first one has electric field intensity within 3 meters lower than certain amount in specific frequency range(approximate transmission range is 5 meters or less in real life scenario), and does not require device manufacture to get certification from administration (but suggested to get ELP certificate for qualifying device won't makes trouble to users but only for company registered in Japan to apply).
[image]
This does not acquire license to operate, no matter what frequency, what kind transmission modulation (e.q. AM/FM) and what usage.
ref: [https://www.tele.soumu.go.jp/e/ref/material/rule/index.htm](https://www.tele.soumu.go.jp/e/ref/material/rule/index.htm)
Second one, does required "Technical Conformity Mark" and which Flipper Zero already have but Flipper Zero only has record on 312~315 MHz, and did not apply all other frequency.
ref: [https://www.tele.soumu.go.jp/e/adm/system/ml/small/index.htm](https://www.tele.soumu.go.jp/e/adm/system/ml/small/index.htm)
These kinds of RF remote controller/devices are commonly in Japan.
ref(ELP certificate list): [http://www.jaama.gr.jp/bijaku/summary/product-list.html](http://www.jaama.gr.jp/bijaku/summary/product-list.html)
ref(ELP sample vendor product manual): [http://www.kato-denki.com/download/manual/img/Bee700_users_KD.pdf](http://www.kato-denki.com/download/manual/img/Bee700_users_KD.pdf)
ref(Specified Low-Power Radio Station sample vendor product page): [https://www.nomura-e.co.jp/products_ts02E_NH_PG.html](https://www.nomura-e.co.jp/products_ts02E_NH_PG.html)

skotopes, 2022-10-16:
Not only Japan but also Canada got same exception. We'll implement it at some point.

skotopes, 2024-01-03:
After some additional thoughts and emails with lab and certification authority we've decided not to implement exception. But we'll keep this in mind when design and certify next generation devices.

closed 2024-01-03 by skotopes

MagicalTux, 2025-01-25:
@skotopes At least an exception for 426Mhz which is the widest standard for remotes here (Japan) would help a lot

skotopes, 2025-01-25:
@MagicalTux we already certified in Japan with very specific conditions. There is no way we can one more band without doing one more certification cycle